foldynl/QLog #240: segmentation fault on ubuntu pacakge

muihlinn, 2023-08-22:
Ubuntu 22.04.3 LTS
X11

clean install from ppa 0.27

got this the first time: 

19:28:30.425 [WARNING ] [0x7f7736a67400]                                    [             ] => Cannot prepare select statement [long int LogFormat::runExport():logformat/LogFormat.cpp:764] 
Segmentation fault (core dumped)

subsequent attempts just printed segmentation fault.

Oddly enough is running fine on a laptop with the previous LTS.

muihlinn, 2023-08-22:
here is the logfile. [logfile.txt](https://github.com/foldynl/QLog/files/12412648/logfile.txt)

Ill attach the coredump later, looks like place has changed in this ubuntu release

foldynl, 2023-08-23:
Thanks.
What I found out:

1) I have performed some tests and I cannot reproduce the issue. My setup is: Ubuntu 22.04.3, XOrg and Wayland, fresh QLog from ppa
2) Unfortunately, I cannot see more details from the logfile and crash file. I can't analyze the coredump because I forgot to generate external symbols in the release mode - this must be fixed in the next version. The logfile points to the error point, but it doesn't show the reason for the crash.

I am attaching the modified qlog binary. It is the same version as you have but the binary contains debug information. Please, could you download this binary, gunzip it, backup the old QLog binary and replace it in /usr/bin/ folder. Then start QLog again. We will see whether it crashes again. If it crashes then please, send me the crash file.
[qlog.gz](https://github.com/foldynl/QLog/files/12416534/qlog.gz)

muihlinn, 2023-08-23:
here it is: 
[_usr_bin_qlog.1026.zip](https://github.com/foldynl/QLog/files/12417214/_usr_bin_qlog.1026.zip)

foldynl, 2023-08-23:
I am adding the backtrace from coredump - just for tracking the issue

```
#0  0x00007fde934052c0 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007fde9340168c in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007fde93401b03 in QHeaderView::restoreState(QByteArray const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00005575c8217b7b in LogbookWidget::LogbookWidget(QWidget*) (this=0x5575cad451b0, parent=<optimized out>) at ../QLog/ui/LogbookWidget.cpp:135
#4  0x00005575c823c24e in Ui_MainWindow::setupUi(QMainWindow*) (this=0x5575caaf78c0, MainWindow=MainWindow@entry=0x7ffe3893e2b8) at ./ui_MainWindow.h:302
#5  0x00005575c822bf85 in MainWindow::MainWindow(QWidget*) (this=0x7ffe3893e2b8, parent=<optimized out>) at ../QLog/ui/MainWindow.cpp:49
#6  0x00005575c8121a28 in main(int, char**) (argc=1, argv=<optimized out>) at ../QLog/core/main.cpp:381
```
That doesn't make sense to me. The error is in the place where QLog restores the geometry of the windows. This place is something that hasn't changed for a long time and QLog goes through it every start.

I have these questions
1) Did you migrate your home from another computer?
2) Could you please give me the file ~/.config/hamradio/QLog.conf Don't worry, there are no passwords
3) You can try running qlog with the parameter -n test: `qlog -n test`
Does the log start or does it crash again?
4) Have you ever tried QLog with QT6?

Sorry for asking so many questions, but it really doesn't make sense to me

muihlinn, 2023-08-23:
Now it's solved because I deleted all the databases and related files, but forgot about .config/hamradio/[...], just deleted it and it went ok.

I'll answer to the relevant questions as it may help someone else in the future.

>     1. Did you migrate your home from another computer?

No,  it was a fresh install (except what it was noted above).
 
>     4. Have you ever tried QLog with QT6?

In fact one of my rigs (sold it recently) was in need of some tweaks in hamlib, so I compiled it, and built qlog from source, my hamlib version and QT6 (but catched some troubles with the online maps). Despite I erased everything I thought it was all from the system and installed from ppa, then trouble began.  

> Sorry for asking so many questions, but it really doesn't make sense to me

Thank you, I was wondering about a config file somewhere.

closed 2023-08-23 by muihlinn