foldynl/QLog #252: QSO Detail / Edit dialog / "My Station"

dl2ki, 2023-11-05:
I have noticed that in the QSO Detail / Edit dialog in the "My Station" tab there is a field called "Operator Name", but it is empty.

[image]
 

The field "Operator Name" is not available in the "My Station" tab in the main window.

[image]

The operator name is already entered in the selected profile in "Settings / Profiles" and could therefore be displayed in the "QSO Detail / My Station" tab. 

[image]

Is there a specific reason why the two "My Station" tabs have different content? But perhaps there is another consideration regarding the content of the tabs.

foldynl, 2023-11-06:
>The field "Operator Name" is not available in the "My Station" tab in the main window.
> The operator name is already entered in the selected profile in "Settings / Profiles" and could therefore be displayed in the "QSO Detail / My Station" tab.

that's according to my proposal. The name and other parameters are defined by the profile and there is no need to display them.

But there is an issue that the QSO Detail Dialog does not show the name of the operator. This must be fixed.

> Is there a specific reason why the two "My Station" tabs have different content? But perhaps there is another consideration regarding the content of the tabs.

Yes. there is. As I wrote, QLog saves these parameters from profile. But if you want to change them later, there must be some way to do it. And that's why there is a different layout for QSO Detail, where you can modify the values independently of the profile.

dl2ki, 2023-11-06:
OK, thank you. Then I can understand that.

commit e48610a5 (2023-11-10): Fixed #252 - QSO Detail Operator Name containes an incorrect value
closed 2023-11-10 by foldynl