foldynl/QLog #253: Calls with VE, VA are coding as Amsterdam & St Paul Islands instead of Canada

willcohen, 2023-11-06:
In my log book, my Canadian entries are repeatedly being looked up and resolving as the wrong country. Is there a database I should be replacing or editing accordingly

foldynl, 2023-11-06:
Please, could you provide me more information? 

1) is your report related to new QSO input or import QSO of QSO Detail (QSO double click)?
2) Please, could you provide me with an example of callsign(s) that is(are) incorrectly identified?
3) QLog version and platform

>  Is there a database I should be replacing or editing accordingly

QLog periodically downloads prefixes and countries (CTY.dat). There is an option to fix records already entered into the database via the QSO Detail Dialog (double click on the QSO). There is also the possibility to fix the records for mapping prefixes to country, but I would not recommend this because QLog updates these records regularly every 21 days.

The first thing is to identify the issue

willcohen, 2023-11-06:
Thank you for the response!

And yes -- these came from importing an ADIF from WSJT-X (before I knew how to having QLog listen), using macOS QLog 0.29:

```
ADIF Export
<adif_ver:5>3.1.1
<created_timestamp:15>20231026 181756
<programid:6>WSJT-X
<programversion:5>2.6.1
<eoh>
<call:6>VE3XFD <gridsquare:4>EN82 <mode:3>FT8 <rst_sent:3>-15 <rst_rcvd:3>-13 <qso_date:8>20231026 <time_on:6>192800 <qso_date_off:8>20231026 <time_off:6>193146 <band:3>20m <freq:9>14.074860 <station_callsign:4>W1WK <my_gridsquare:6>FN42KG <eor>
```

Entries detected by listening to the WSJT-X server aren't having this issue.

Even when I manually adjust the value of the country by right clicking on the callbook, the double-clicked value still reads as Amsterdam & St Paul Islands:

[image]

foldynl, 2023-11-06:
Many thanks for the example above.

where do you see this wrong country ? I can confirm the wrong behavior in the QSO Detail (the dialog you sent above). But otherwise I see Canada in the export ADI, Log Table etc. 

Could you list everything where you can see the wrong value?

willcohen, 2023-11-06:
Excellent question. It took me a minute to figure it out after you asked --

I made a new fake QSO to not mess up the existing entry:

```
ADIF Export
<adif_ver:5>3.1.1
<created_timestamp:15>20231026 181756
<programid:6>WSJT-X
<programversion:5>2.6.1
<eoh>
<call:6>VE3AAA <gridsquare:4>EN82 <mode:3>FT8 <rst_sent:3>-15 <rst_rcvd:3>-13 <qso_date:8>20231105 <time_on:6>192800 <qso_date_off:8>20231105 <time_off:6>193146 <band:3>20m <freq:9>14.074860 <station_callsign:4>W1WK <my_gridsquare:6>FN42KG <eor>
```

When I import that ADI in, it shows up correctly as Canada in the main log window, but then the double-clicked detail shows Amsterdam & St Paul. If, from that detail window, I then edit and change any other details on the entry, upon saving, it then saves it with the incorrect country. To be clear, when saving, it does note that the country and continent appear incorrect, but I'd have to manually change the country every time I save to avoid the country being re-saved incorrectly.

[image]

foldynl, 2023-11-06:
the good news is that I am able to replicate everything you described above. The bad news is that I still don't see what's causing the issue. Currently, it seems that this is an issue of QSO detail dialog only.

Quick work is

1) Do not modify QSO via the QSO Detail dialog
2) you can directly fix the values.  Select all impacted values in the Log table, then the context menu for the Country column (click the right mouse button), select "Edit Value",  change value to "Canada" and press enter. All selected records will be changed to the new value "Canada" as you can see below. 

This procedure should not be used for common editing QSOs because it only checks a very small set of field dependencies.

[image]

Otherwise, thanks for reporting the error.

willcohen, 2023-11-06:
Thank you for diagnosing! And thank you for your work on this tool -- it is extremely helpful to have an active open-source cross-platform program that works on macOS and provides so much functionality.

commit e7c020d4 (2023-11-10): Fixed #253 - Calls with VE, VA are coding as Amsterdam & St Paul Islands instead of Canada

All model.match calls were changed from default Qt::MatchStartsWith|Qt::MatchWrap to Qt::MatchExactly
(DXCC of Amsterdam is 10 and Canada is 1 but Amsterdam is before Canada in alphabetical order)
closed 2023-11-10 by foldynl