foldynl/QLog #257: 0.29.1 Macosx Sonoma - QLog is already running error popup

jrjesson, 2023-11-11:
After upgrading to 0.29.1 I now receive an error popup "QLog is already running" when starting QLog.  Repro'd on an M1 Macbook Pro , and an M2 Mac Mini. The error does not occur on 0.29.0.  I cloned 0.29.1 and built from scratch to confirm the error.

foldynl, 2023-11-11:
it is strange because this release does not touch this part of QLog. Unfortunately, I am not able to test it because MacOS build is only prepared by GitHub - I have no MacOS hardware. The only thing that can be wrong is the version of the Qt library, because as I looked, QLog 0.29.1 uses the latest release (6.6), which can be quite problematic. 

Please, could you  send me the version of Qt that 0.29 is built with (Screenshot of Help->About)? Unfortunately, I don't have access to the build log on GitHub anymore for release 0.29.

foldynl, 2023-11-11:
sorry, maybe a useless question, but are you sure that QLog is not really running? Sometimes it happened that QLog was no longer active, but one of the QLOG processes remained active (there was an issue with a Chrome process - QLog's online map).

jrjesson, 2023-11-11:
Yes, I was scratching my head about this as well, I did a ps to confirm no running instances, and tested on my two macs. I did a diff on the code between 0.29 and 0.29.1 and no change at all around .AppGuard or main.cpp  When I compiled I did just pull QT 6.6 via homebrew to do the build so it could be an issue with that QT release.  I'll research to see if I can compile with an older version of QT to confirm. Can you tell me what version of QT you used on 0.29.0?

foldynl, 2023-11-11:
That's a question I can't answer. Please try to download the DMG for 0.29.0 from the GitHub Release section, after launch go to Help->About and you will see the Qt version there. Unfortunately, I don't have MacOS HW, so I can't do this - I use the services of GitHub to prepare all DMGs.

I have tested the QLog with Qt6.6 under Windows and there is no issue with the AppGuard. Maybe, it is specific issue for MacOS.

foldynl, 2023-11-11:
I extracted the DMG under Linux and it seems that 0.29.0 was prepared based on QT 6.5.2

jrjesson, 2023-11-11:
It was going to be a little involved to downlevel QT6, homebrew doesn't appear to support that easily - it only allows installing QT5 or QT6.  

So, I wondered instead if QT6.6 might have fixed or changed something.  I took a deeper look into AppGuard.cpp and noticed the following:  In the method ::IsAnother Running(), the current logic:

`bool AppGuard::isAnotherRunning(void)
{
    FCT_IDENTIFICATION;

    **if ( sharedMem.isAttached() )
        return false;**

    memLock.acquire();
    const bool isRunning = sharedMem.attach();

    if ( isRunning )
    {
        sharedMem.detach();
    }

    memLock.release();

    return isRunning;
}
`
That first check for shared memory attachment seems (asterisked/bolded) like that would always return false on startup since (I think) the shared memory segment wouldn't have yet been attached to the new process -- that seems to happen later in the ::tryToRun() method.  I commented out the isAnotherRunning() check in ::tryToRun and the app startup behavior is as intended - the first instance of QLog runs, the second fails because this call fails: const bool result = sharedMem.create( sizeof( quint64 ) );  So the popup is provided to subsequent starts. 

I'm unsure how to suggest refactoring of the ::isAnotherRunning() method, it seems unnecessary given the other checks, but I might be missing a peculiarity of other platforms?  Here's the code I commented out: 
`bool AppGuard::tryToRun(void)
{
    FCT_IDENTIFICATION;

//   if ( isAnotherRunning() )   // Extra check
//   {
//        return false;
//   }

    memLock.acquire();
    const bool result = sharedMem.create( sizeof( quint64 ) );
    memLock.release();
    if ( !result )
    {
        release();
        return false;
    }

    return true;
}`

foldynl, 2023-11-12:
Source: https://doc-snapshots.qt.io/qt6-dev/whatsnew66.html#qt-quick-module

_The new [QNativeIpcKey](https://doc-snapshots.qt.io/qt6-dev/qnativeipckey.html) class holds a native key used by [QSharedMemory](https://doc-snapshots.qt.io/qt6-dev/qsharedmemory.html) and [QSystemSemaphore](https://doc-snapshots.qt.io/qt6-dev/qsystemsemaphore.html), providing better support for sandboxed applications on Apple platforms. See [Native IPC Keys](https://doc-snapshots.qt.io/qt6-dev/native-ipc-keys.html) for more information._

it seems that something has really changed in QT6.6. Currently, I don't know what exactly this means for our case, but it is obvious that they changed something there.

foldynl, 2023-11-12:
Please, could you test the patch above (branch bugfix-257)?

jrjesson, 2023-11-12:
To your earlier question 0.29.0 used QT 6.5.2.  I tested a bit yesterday when I commented out  ::isAnotherRunning() the app started and behaved as intended - when I started another instance of the app that second instance exited with "QLog is already running." Pulling the change to test now.

jrjesson, 2023-11-12:
I pulled and compiled from your test branch and the popup error still occurs.  Curious, I wanted to see what the error result was on the failing call in ::tryToRun 
`- const bool result = sharedMem.create( sizeof( quint64 ) );.  `
So I added an fprintf - 
`
        // JRJ: print out the error
        fprintf(stderr, "\nError = %d %s\n\n", sharedMem.error(), sharedMem.errorString().toLocal8Bit().data() );
        release();
`
The result is the following: 

*Error = 5 QSharedMemory::attach (shmget): doesn't exist*

jrjesson, 2023-11-12:
Seeing that prior error, and noting the lack of an .attach()  call in the sequence before the create, I added one to the ::tryToRun() method.  This seems to fix the issue.  I'm not experienced in QT development so am unsure if perhaps QT6.6 made it required to do the .attach() before the .create()?  

bool AppGuard::tryToRun(void)
{
    FCT_IDENTIFICATION;

    if ( isAnotherRunning() )   // Extra check
    {
        return false;
    }

    memLock.acquire();
    //JRJ: Add an attach call before the create
    sharedMem.attach();
    const bool result = sharedMem.create( sizeof( quint64 ) );
    memLock.release();
    if ( !result )
    {
        // JRJ: print out the error
        fprintf(stderr, "\nError = %d %s\n\n", sharedMem.error(), sharedMem.errorString().toLocal8Bit().data() );
        release();
        return false;
    }

    return true;
}

foldynl, 2023-11-12:
it's a mystery why it works. I have tested it on linux and your attach has no negative effect on the function of the class. 

Please, could you add qInfo()<< before your attach and test it again? What does it return for the first instance and the second instance of QLog?
```
if ( isAnotherRunning() )   // Extra check
{
    return false;
}

memLock.acquire();
//JRJ: Add an attach call before the create
qInfo() << sharedMem.attach();
const bool result = sharedMem.create( sizeof( quint64 ) );
memLock.release();
if ( !result )
{
    // JRJ: print out the error
    fprintf(stderr, "\nError = %d %s\n\n", sharedMem.error(), sharedMem.errorString().toLocal8Bit().data() );
    release();
    return false;
}

return true;
```

jrjesson, 2023-11-12:
Ha, this behavior was bugging me so I was reading the docs when you replied.  I added qinfo, it returns the following:
`10:19:43.908 [INFO    ] [0x1da589ec0]                                    [             ] => false [bool AppGuard::tryToRun():core/AppGuard.cpp:109] `

That didnt seem completely helpful, so I added another fprintf to get the error and error code - it returned 
`didAttach 5 QSharedMemory::attach (shmget): doesn't exist`

To your earlier point, this feels like a bug in QT 6.6/macosx, the create call is supposed to have an implicit attach, so...

foldynl, 2023-11-12:
Many thanks for your assistance. To understand this, we would have to dive into QT source code....OK... I'll add the extra `attach` with a note that this is a hack for QT6.6. In the meantime, I have tested it on other platforms and QT versions and it has no impact no function. So it is not necessary to add an extra compilation condition there. 

I will try to publish it as a fix, but unfortunately I don't have time to do it today. I will try to create a new release tomorrow, so the fixed DMG could be available tomorrow.

jrjesson, 2023-11-12:
No worries on timing, I have my build :-)  I did send an email based on your github profile for any future followups.

commit 71dde90c (2023-11-13): Fixed #257 - QLog is already running error popup - attach workaround

Many thanks to jrjesson for the analysis and the patch.
commit 67f34e56 (2023-11-13): Fixed #257 -  QLog is already running error popup
closed 2023-11-13 by foldynl