foldynl/QLog #259: QLog start graphic

dl2ki, 2023-11-11:
QLog v 0.29.1 , Kubuntu 22.04, KDE-Plasma 5.24.7
Graphics Nvidia-Driver 470

---

The graphic when starting QLog is displayed with a black background.

[image]

I don't have an older version of QLog here for a test, but I think that the graphic was previously displayed with a transparent background.

foldynl, 2023-11-11:
I am not able to replicate this under any of my test instances (Ubuntu 22.04 - Nvidia/Intel with Wayland or Xorg, Kubuntu 22.04 Software Virtual Render, Ubuntu 22.10 Software Render, Windows 10 AMD Zen). But maybe I'll try to fix it in next release, because I found a message on the QT forum about a similar issue.

foldynl, 2023-11-11:
Please, would it be possible to test the attached binary? It is the same QLog release as current (0.29.1) + two new attributes for Splash Screen. 
The testing sequence should be:

1) copy the gzip file to temp folder and ungzip it (gzip -d qlog_splash_test.gz)
2) chmod u+x qlog_splash_test (if it is needed)
3) ./qlog_splash_test

Does it helps?

[qlog_splash_test.gz](https://github.com/foldynl/QLog/files/13326156/qlog_splash_test.gz)

dl2ki, 2023-11-11:
No change. The same effect occurs.

The installation of an older version "0.29.0" or "0.28" will probably not be possible because of the database? The problem may also have been caused by a regular update or one from the "kubuntu-backports" PPA.

I haven't noticed anything with other programs so far.

foldynl, 2023-11-12:
Downgrade to 0.29.0 is possible because release 0.29.1 is only bugfix release - no change to database. I have tested Nvidia-Driver 470 without any issue - by the way, newer Nvidia driver is available in Ubuntu (NVidia Driver 535).

Are you seeing any other issues like this after start or is it just this splash screen?

dl2ki, 2023-11-12:
Is it possible to download the 0.29.0 Debian package from your PPA?
Unfortunately, I no longer have the package here.

foldynl, 2023-11-12:
You can directly download it via this link.

https://launchpad.net/~foldyna/+archive/ubuntu/qlog/+files/qlog_0.29.0-1~jammy_amd64.deb

Keep in mind that it is needed to temporary block the ppa with QLog to prevent unwanted upgrade.

But to be honest, I think that this issue is not related to QLog 0.29.1

dl2ki, 2023-11-12:
You are right. Same behavior.

Unfortunately, I can't contribute anything else to the solution. People often don't pay attention to the start graphic.

In my Logwatch logs I could see an update of the Nvidia drivers from 4.11.2023.
(libnvidia-cfg1-460:amd64 470.199.02-0ubuntu0.22.04.1 => 470.223.02-0ubuntu0.22.04.1) and others.
Otherwise there is nothing conspicuous to be seen here.

foldynl, 2023-11-12:
I'm afraid I'm not able to provide a fix at the moment. It is minor issue. it's an issue with minimal impact. I leave this issue open.

please close this issue when you see a positive change.

foldynl, 2023-12-02:
Wolfgang, thanks for the info. I'm glad that's clear now. I saw a similar issue, but then it was an issue with VMware 3D Driver for Linux. So I was hoping it would be something similar in your case.

closed 2023-12-02 by foldynl