foldynl/QLog #261: RST fields revert to 59 after changing them, then tabbing to next field

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
Having entered a callsign, then using tab to move into, and set the RST fields, and then tabbing out to the next field, the RST fields change back to 59 and 59, regardless of what was entered.

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
Linux  6.1.62-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Nov  9 03:01:44 UTC 2023 x86_64 GNU/Linux

With the dark mode switched on in case that is relevant

foldynl, 2023-11-15:
Do both fields change or only one of the RST fields?

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
Both

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
Sorry accidentally hit the close button, my apologies.

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
[image]

Actually may be time based or random, rather than on tabbing...

foldynl, 2023-11-15:
sorry to have so many questions, but it doesn't make any sense to me even when I see it. It is obvious that RST fields were deleted before the tabbing.

Is your rig connected to QLog? 
Did you have WSJT running when you shot this? 
Has this appeared before, or is this the first time you've seen it? 
Have you used any versions of QLog before with same issue or is this your first on this computer?
I guess, you are running QLog 0.29.x. 
Is it installed from package or you compiled it from source code?

c-o-m-m-a-n-d-e-r-d-a-t-a, 2023-11-15:
Yes, agree that it's not the actual tabbing, but happens by itself around the same time.
Rig is connected, via wfview in between.
No WSJT.
I don't remember seeing this before, but it's possible my usage pattern has changed...
Qlog 0.29.0
Compiled from git.

Actually I tested, it's the rig connection. If I disconnect rig, the behaviour stops. If I connect rig mid-qso, it instantly jumps back to 59/59

foldynl, 2023-11-16:
thanks for the detailed description. Now it's up to me to do the research. It is interesting because the QLog's RST fields are not dependent on any event that would occur during the QSO. They depend on changing the Mode, Frequency (partly) and Callsign. If it is observed during a QSO, then it may be some kind of external event.

foldynl, 2023-11-17:
I found a potential issue. 

QLog changes RST values when RIG Mode is changed. I also consider a change of passband width (filter width) to be a change of the rig mode. 

I don't know why, this is actually a question for Elliott, but wfview changes the bandwidth without an external event, as you can see below the mode remains unchanged but the filter changed from 2400 to 3600 (see especially the timestamp, how fast this event is generated)  . I will try to fix this in the next QLog.

```
15:52:17.391 [DEBUG   ] [0x7f169bc8e400]        [qlog.ui.newcontactwidget.function.entered] => *** [void NewContactWidget::changeModefromRig(VFOID, const QString &, const QString &, const QString &, qint32):../QLog/ui/NewContactWidget.cpp:835] 
15:52:17.391 [DEBUG   ] [0x7f169bc8e400]     [qlog.ui.newcontactwidget.function.parameters] => "SSB"   "USB" [void NewContactWidget::changeModefromRig(VFOID, const QString &, const QString &, const QString &, qint32):../QLog/ui/NewContactWidget.cpp:837] 
15:52:17.391 [DEBUG   ] [0x7f169bc8e400]        [qlog.ui.newcontactwidget.function.entered] => *** [void NewContactWidget::__modeChanged(qint32):../QLog/ui/NewContactWidget.cpp:774] 
15:52:17.391 [DEBUG   ] [0x7f169bc8e400]     [qlog.ui.newcontactwidget.function.parameters] => 2400 [void NewContactWidget::__modeChanged(qint32):../QLog/ui

QString &, const QString &, qint32):../QLog/ui/NewContactWidget.cpp:835] 
15:52:17.932 [DEBUG   ] [0x7f169bc8e400]     [qlog.ui.newcontactwidget.function.parameters] => "SSB"   "USB" [void NewContactWidget::changeModefromRig(VFOID, const QString &, const QString &, const QString &, qint32):../QLog/ui/NewContactWidget.cpp:837] 
15:52:17.932 [DEBUG   ] [0x7f169bc8e400]        [qlog.ui.newcontactwidget.function.entered] => *** [void NewContactWidget::__modeChanged(qint32):../QLog/ui/NewContactWidget.cpp:774] 
15:52:17.932 [DEBUG   ] [0x7f169bc8e400]     [qlog.ui.newcontactwidget.function.parameters] => 3600 [void NewContactWidget::__modeChanged(qint32):../QLog/ui

```

dl2ki, 2023-11-17:
Yes, changing the filter bandwidth sets RSTs and RSTr back to 599. It often happens that you adjust the filter bandwidth during a QSO. This could be one reason for my observations, but I'm not sure. But I will pay attention to it now.

closed 2023-12-01 by foldynl