foldynl/QLog #263: pfx field

ik3ocd, 2023-11-17:
good morning from flavio ik3ocd these days I'm trying your qlog work and a nice project, I noticed that the import from other logs is successful and all the fields are ok, but upon a new QSO insertion the information for the prefix in the pfx field seems to be missing , perhaps not yet implemented, the pfx information on QSO insertion is also missing. Thanks for any response on this. 73'

my kde 39 operating system
​
[image]

foldynl, 2023-11-17:
Currently, It is not yet implemented. QLog only accept the PFX from external sources (WSJTX, import etc).

But it is easy to add it. It will be a part of next release.

ik3ocd, 2023-11-17:
grazie foldynl ,Thanks for your reply, I like your work and I look forward to further qlog progress. ciao e 73 de ik3ocd flavio

closed 2023-12-01 by foldynl