foldynl/QLog #264: QLog Flatpak

AsciiWolf, 2023-11-18:
I am working on a [Flatpak](https://flatpak.org) package

AsciiWolf, 2023-11-18:
https://github.com/flathub/flathub/pull/4709

foldynl, 2023-11-18:
Many thanks for the Flatpak.

I have no previous experience with flathub. I will try to study it. Please grant me the rights. I will probably need your help from the beginning. 

I went through your XML and the only thing that might be a little inaccurate is the link to the image. I would use [non-permanent link](https://foldynl.github.io/QLog/screens/qlog_main.png) instead of [the permanent link](https://camo.githubusercontent.com/03156037d0bfbcf17a1d34ef43e170fe28aa4de0178cb6ddf4e1a15718bee0ff/68747470733a2f2f666f6c64796e6c2e6769746875622e696f2f514c6f672f73637265656e732f716c6f675f6d61696e2e706e67) . If I update the image in the repo, this link will be always the correct image.

I will read the flatpak documentation, which is all necessary for the new release. And I have to update my large a list of QLog release instructions (QLog is prepared for 4 versions of Ubuntu, 2-3 versions of Fedora, MacOS, Window and now also for Flatpack)

foldynl, 2023-11-19:
Please, may I kindly ask you for the following information?

1) Is any action expected from my side regarding publishing on flathub?

2) sorry, I'm quite busy, so I only briefly read the Flatpak documentation. Could you please write me in a simple list what needs to be done in case of a new release? I saw that the commit hash is a part of the Flatpak yaml. So it will probably be necessary to do a new PR for each new release. Am I correct?

3) Will it be okay if I merge your QLog's PR in the next release? Or is it necessary to be a part of repo before acceptation Flathub's PR?

AsciiWolf, 2023-11-19:
1. Nope, nothing needed from your side. Just accept the push rights invitation after the QLog repo gets created on Flathub.
2. We have something called "external-data-checker" on Flathub that automatically checks for new releases (GitHub tags etc.) and create Pull requests that update the Flatpak manifest file. So, ideally just downloading/installing and quick testing (I can do that if you wish) of the test build that will automatically get created in such PR could be sufficient. If there will be a problem, I can handle it or give you a hint, no worries.
3. Sure, feel free to include it in the next release. I have added the changes from this PR as patches directly to the Flatpak for now.

foldynl, 2023-11-23:
Many thanks.

Closing the issue. Issues related to flatpack will be reported directly via [flathub/io.github.foldynl.QLog ](https://github.com/flathub/io.github.foldynl.QLog/issues) repository

closed 2023-11-23 by foldynl