foldynl/QLog #267: Some things I noticed in Linux

lu9dce, 2023-11-22:
1) The online map does not work, the screen is blank.

2) Importing ADI takes a very long time, especially with an ADI of 10,000 QSOs (6 months of work).

3) Exporting ADI generates an ADI in a LOTW-style format, but it should be in a WSJT-X format.

4) It would be helpful to make it easier or more visible to create a new ADI.

Qlog is a great effort and is appreciated.

Something that could be added is that, when connected to a cluster, adding a new QSO would generate a spot in the cluster.

Thank you for your hard work.

Best regards from LU9DCE

foldynl, 2023-11-22:
Thank you for your feedback. To move forward, I would need some information. See comments below.

> 1. The online map does not work, the screen is blank.

Please, could you share your setting? Is QLog Installed from an installation package (deb, rpm, flatpak) or from source code? CPU type (amd64, aarch64)? OS distribution?

>2. Importing ADI takes a very long time, especially with an ADI of 10,000 QSOs (6 months of work).

"Very long time" is subjective information. Can you please specify the time?

Yes, QLog has performance issue in some area. Any help is welcome

 
> 3. Exporting ADI generates an ADI in a LOTW-style format, but it should be in a WSJT-X format.

The ADIF specification does not define any LOTW/WSJTX style. Therefore I do not see any reason to change ADI-style in QLog.
 
> 4. It would be helpful to make it easier or more visible to create a new ADI.

I'm sorry, but I don't understand you. What does "create a new ADI" mean?

> Something that could be added is that, when connected to a cluster, adding a new QSO would generate a spot in the cluster.

See https://github.com/foldynl/QLog/wiki/DX-Cluster

Maybe `Spot Last QSO` is the command what you are looking for.

> 
> Thank you for your hard work.
> 
> Best regards from LU9DCE

lu9dce, 2023-11-22:
OK my distro 

Linux elemental.slackware 5.15.117 #1 SMP PREEMPT Thu Jun 22 15:39:49 CDT 2023 x86_64 Intel(R) Core(TM) i3 CPU         530  @ 2.93GHz GenuineIntel GNU/Linux

Slackware 15.0

I compiled Qlog from the source code

> I'm sorry, but I don't understand you. What does "create a new ADI" mean?

I mean being able to create different adi profiles
and when deleting these profiles the addi is deleted
to work to be able to work with different licenses and different duty books
and its conventions, that is, 2 stations can operate the same book or separately.

> See https://github.com/foldynl/QLog/wiki/DX-Cluster
> 
> Maybe Spot Last QSO is the command what you are looking for.

something like that is fine but automatically
If it is connected to a cluster and the qso was performed, the spot will be created automatically

-----------

What I did not check if when performing the qso at the time of saving it is uploaded to the online guard books automatically!

--------------

I would like to be able to help with the optimization but my knowledge of C or other languages other than PHP is very limited.

I have been programming in PHP for many years

Processing an add of several megabytes in PHP takes a few seconds, other languages that can compile would have to go faster

----------

I have published 2 programs made in PHP on GitHub
one is an autospot and another a bot for wsjtx or jtdx

Autospot can automatically upload many online guard books

Maybe if you look at the modules part, even if they are in PHP, they can give you some ideas to add to qlog
so you can upload to CLOUDLOG and Argentina Log

autospot can also create an email with a qsl attached and send it

https://github.com/lu9dce/autospot

-------

thank you

Excuse my English, I'm using a translator

foldynl, 2023-11-22:
> Linux elemental.slackware 5.15.117 #1 SMP PREEMPT Thu Jun 22 15:39:49 CDT 2023 x86_64 Intel(R) Core(TM) i3 CPU 530 @ 2.93GHz GenuineIntel GNU/Linux
> 
> Slackware 15.0
> 
> I compiled Qlog from the source code

And how long does it take to import 10k QSO?

> I mean being able to create different adi profiles
> and when deleting these profiles the addi is deleted
> to work to be able to work with different licenses and different duty books
> and its conventions, that is, 2 stations can operate the same book or separately.

I'm sorry, but I still don't see what is requested. Can you describe it differently? It's enough to describe it really simply in points

> something like that is fine but automatically
> If it is connected to a cluster and the qso was performed, the spot will be created automatically

I definitely don't want to implement this. This approach adds so many useless reports to DXCluster network.  In this mode, operators do not recognize what is DX and what is not DX and send also "local" stations as well.

> What I did not check if when performing the qso at the time of saving it is uploaded to the online guard books automatically!

Everything is described in [QLog's WIKI](https://github.com/foldynl/QLog/wiki). Shortly. QSO are NOT automatically uploaded to Online logbooks because each online service has a different QSO management. And, for example, DELETE is implemented differently for every implemented service - for example, for LoTW it is not recommended to send QSO online, but only in batches. QRZ has a special QSO DELETE procedure. That's why I leave it up to the operator when he/she sends a QSO to online service - when he/she is satisfied with the QSO record.

Other problems arise in case of unavailability of one or more online services. You have to store the QSOs and flush them when become online. To be hones, it is not easy to manage it properly, if you have two or more service.

> I have published 2 programs made in PHP on GitHub one is an autospot and another a bot for wsjtx or jtdx
>
> Autospot can automatically upload many online guard books
>
> Maybe if you look at the modules [...]

QLog generates the  ([Network Notification Messages](https://github.com/foldynl/QLog/wiki/Notifications)). If you want implement a bridge for sending an online messages, you can use the messages.

One comment to your autospot. I didn't study it in the details, but it seems to me that Update and Delete are not solved for  services (for example [QRZ](https://github.com/lu9dce/autospot/blob/953c61142ca8e06e5bb7f0cc3206a77f3d24ddcb/web/modulos/qrz.php#L15)). How is repeat solved when an online service report an error? etc. Based on the Cloudlog's Wiki, Cloudlog doesn't have the Delete API , I'm not sure about Update API etc. How does autospot solve the cases when the operator modifies or deletes the QSO in a log? 

How I said, It is not easy to properly implement "Immediately send QSO" service.

lu9dce, 2023-11-22:
> One comment to your autospot. I didn't study it in the details, but it seems to me that Update and Delete are not solved for services (for example [...]

It does not.. the contact once made is uploaded to the online guard books.. autospot is not a guard book itself.

it processes the qso made and sends it, it is an ultron plugin that makes the qso automatically, I have had both programs running with jtdx in slackware for more than 5 years without intervention, all the qso that I make are made by ultron and autospot uploads them and send email to the correspondent

foldynl, 2023-11-23:
ok, back to the QLog topic. I take these things from the discussion related to QLog.

1) The online map does not work, the screen is blank.
Currently I cannot replicate it and no other user reported this. I have to install Slack 15 and test it.

2) Importing ADI takes a very long time.
Unfortunately, you didn't write how long it took, but I take it as a known issue that needs to be worked on

To make the situation clearer, I would close this issue report and open a separate issue reports for the mentioned things above.

closed 2023-11-23 by foldynl