foldynl/QLog #268: 'detached' window won't 'attach'

foldynl, 2023-11-22:
Tested and it seems that impacted platforms are QT 5.1x under wayland and Flatpak under wayland.

Temporary solution described in the discussion #266 

### Discussed in https://github.com/foldynl/QLog/discussions/266

<div type='discussions-op-text'>

<sup>Originally posted by **g7kse** November 21, 2023</sup>
Probably something really daft but I managed to detach the dx cluster window and went to re attach it and can't seem to do it. Now its floating as a second window. 

I know its something really simple but I've had it floating for days now and can't seem to get it back...please someone put me out of my shame :-)

[image]
</div>

foldynl, 2023-11-23:
flatpak is prepared without Wayland support. https://github.com/flathub/io.github.foldynl.QLog/pull/4

g7kse, 2023-11-24:
Sorted, works with flatpak thanks

foldynl, 2023-11-24:
excellent.  

it seems Wayland will be a nightmare for me. QT is not properly prepared for it

closed 2023-11-24 by foldynl