foldynl/QLog #271: Cannot change TQSL Path in Settings / Sync & QSL / LoTW

lb7zg, 2023-11-24:
It is not possible to change TQSL Path in  Settings / Sync & QSL / LoTW via the browsing feature. It is not possible to enter path manually either.
Qlog 0.29.2, TQSL 2.6.5, on Debian Bookworm

foldynl, 2023-11-24:
Please, could you share more info? What means "It is not possible to change"? You cannot click on the button or it does not open a "Select File" Dialog, you don't see any TQSL instance there? Animated GIF should also help to understand your issue. 

Please, could you also share where you are using flatpak or deb package.

lb7zg, 2023-11-24:
Hi Ladislav

I'm not allowed to manually change the information in the path-line.
Maybe this is by design? The Select File Dialog opens but when choosing
the right path / another path via the Select File Dialog, the unedible
information in the path-line does not change. When trying to use the
LoTW functionality, an error massege from QLog appears: "Unexpected
Error from TQSL".
I'm using flatpak.

Besides this little bug, the software is very nice and feature rich!
Thank you!
Regards and 73 
Bjorn

& QSL / LoTW (Issue #271)
Date: 11/24/2023 11:16:47 AM

foldynl, 2023-11-24:
Unfortunately, it is an issue associated only with flatpak. At this moment I have no workaround for it. I am soling this  with flathub. More details flathub/io.github.foldynl.QLog#5

just to clarify. is TQSL installed in the same way as flatpak?

lb7zg, 2023-11-24:
OK! Thank you for the feedback and your time.
Yes, TQSL is isntalled via flatpak.

73 Bjorn

& QSL / LoTW (Issue #271)
Date: 11/24/2023 11:05:01 PM

commit cae9366c (2023-11-29): Fixed #271 - Cannot change TQSL Path in Settings

The fix is more complex mainly on the Flatpak manifest site.
closed 2023-12-01 by foldynl