foldynl/QLog #275: Add dynamic load i18n file support

BG7JAF, 2023-11-29:
Hi Ladislav:

I found QLog is a wonderful logging tool, and want to recommend to more Chinese Ham to use it. So I do some translation and generate a qlog_zh_CN.qm file, put it in program install path's "translations" folder, but it dosn't work. Then I read your code and found that the translation files are load from qt resource,  dose that mean I have to recompiled the software to add new translation language support? Many Qt software can support dynamic load language file such as [KLog](https://github.com/ea4k/klog) . Can you add this feature to QLog? thanks

foldynl, 2023-11-29:
thanks for the feedback. There have already been several similar requests, but then I have this opinion.

I would like to keep everything about QLog in one place - github. I will not reject any PR for the translation of any language, but I would like this translation to be part of the source code. I don't want pieces of QLog to be downloaded as plug-ins in different places. Another argument is that Qlog is provided in many type of packaged and OS versions and it is easy to manage it from one place.

it's my current opinion. However, it can change over time, if someone convinces me that it is better to have separate translations.

Yes, I recognize that there is one problem with a new release translations. I solved this problem in the case of the German translation that I have an agreement with Wolfgang (DL2KI). Just before the releasing, when the texts are ready, I will ask Wolfgang to update the German transaction file. This will ensure that the translations are up-to-date when a release is issued.

What prevents the Chinese translation from becoming part of QLog?

BG7JAF, 2023-11-29:
Hi Ladislav:

Thanks for your reply. I suggest  using dynamic loading mainly because of the workflow I'm used to. I like to try to update the software interface immediately after making some changes, because often there are several Chinese words corresponding to the word that needs to be translated (I'm sure there are similar cases in other languages), and the correct translation needs to be selected based on the context of the software interface. If you can load the translation files dynamically, the whole process will be much faster than compile-do some modify-recompile-process. And using dynamic loading does not affect the translation files become part of the project source code, you can still store them on your github.
If the software is not going to make dynamic loading soon, I will consider building a compiling environment to confirm my translation.

foldynl, 2023-11-29:
> If the software is not going to make dynamic loading soon, I will consider building a compiling environment to confirm my translation.

this is the reason why I should think about it. Because, not everyone is willing to create a build environment for translation verification... Thanks.

Unfortunately, if it will be implemented, it will not be in the next release. The next release is almost at the end of the development cycle.

foldynl, 2023-11-30:
maybe it will be part of the next release.

commit 17d79777 (2023-12-01): Fixed #275 - Minor rewording
commit c2f38c6a (2023-12-01): Fixed #275 - Added dynamic load i18n file support
closed 2023-12-01 by foldynl