foldynl/QLog #291: Not possible to enter a Satellite QSO

ea5wa, 2023-12-14:
Is not possible to enter a satellite QSO, because there is no satname and satmode fields (or better with a list)

[image]

foldynl, 2023-12-15:
There are the SAT fields. QLog field layout was changed in 0.29 when Sat-Tab was removed and newly you can define own layout with sat fields via Layout->Edit->Add. The SAT field are `SAT Mode` and `SAT Name`.

ea5wa, 2023-12-15:
Hi Ladislav, thanks for your answer.

I've done it, I've added satname and Satmode to the layout but I can not put focus on these fields and therefore I can't write on them

[image]

foldynl, 2023-12-15:
Did you change `Propagation Mode` to `Satellite`? 

I have the same layout for testing and everything is working. See below.

[image]

ea5wa, 2023-12-15:
Ok. That's the problem....

On the Mode's list I can't see the correct mode for QO-100 satellite. It should be S/X. Is it possible to edit that list?

foldynl, 2023-12-15:
Ok. it is clear to me now. Unfortunately, the list is built-in therefore it has to be fixed.

ea5wa, 2023-12-15:
Thanks 73

foldynl, 2023-12-16:
BTW. It seems that SAT_MODE field is an optional field. It means that it does not have to be filled for a valid QSO. Based on [LoTW](https://lotw.arrl.org/lotw-help/satellite-qsos/?lang=en) and [eQSL](https://www.eqsl.cc/qslcard/eSatelliteRules.cfm) service specification, the valid Sat QSO must contain Propagation mode = 'SAT' and valid Sat Name.

However, it does not change the fact that SX will be added to QLog. I think that correct source of the Sat modes can be this [link](https://www.pe0sat.vgnet.nl/satellite/sat-information/modes/).

ea5wa, 2023-12-16:
Then, are you going to remove that list (satmode)? Or is better to have all the possibilities in that list? From my point of view, for a satellite QSO is interesting to have RX frequency or RX band visible. A satellites dropdown list is useful

foldynl, 2023-12-16:
> Then, are you going to remove that list (satmode)? Or is better to have all the possibilities in that list?
> 
SX will be added to QLog. Sat mode remains as is.

> From my point of view, for a satellite QSO is interesting to have RX frequency or RX band visible. A satellites dropdown list is useful

correct me if I'm wrong, but the QO-100 RX/TX offset has a constant value. This means that you just need to enter the correct RIT/XIT offset for your rig in QLog and you will see the correct frequencies in the QLog. Sorry, I don't have a lot of SAT experience, if there can be a situation where the RX and TX frequency offset will not have a fixed value or it will change non-linearly. In this case, QLog would not be able to record the correct frequency.

ea5wa, 2023-12-16:
I understand RIT/XIT use in the same band, but normally satellites has uplink/downlink in different bands

foldynl, 2023-12-16:
I am sorry. I wanted to say RX/TX Offset in the Rig setting dialog

[image]
 
Otherwise RX/TX freq is displayed when Manual Mode is enabled.

ea5wa, 2023-12-16:
Not useful for sats. Just some of them like GreenCube

foldynl, 2023-12-16:
if we neglect the doppler effect, then it is usable. Yes, SPLIT mode would be better. It is planned.

ea5wa, 2023-12-16:
I always neglect doppler efect. Just simply introduce the frequency (and band) for Uplink and the frequency (and band) for Downlink for that QSO. Normally, I take these frequencies from a satellites list. These images are taken from KLog (the soft I've been using until now):

[image]
[image]

commit 041a44ff (2023-12-16): Fixed #291 - Added missing SX Sat mode
closed 2024-01-05 by foldynl