foldynl/QLog #296: No FT4 mode included

ea5wa, 2023-12-21:
Trying to import some FT4 QSOs from another software, QSO are saved as MFSK mode. I take a look to the Settings-Modes and realized that there are no FT4 mode.

Is it possible to include it in the next release?

Thanks 73

foldynl, 2023-12-22:
FT4 mode does not have a separate mode, but it is a submode of MFSK. That's why it can't be seen in Settings-Modes. But if you activate MFSK, you can see it as a its submode.

Getting back to import, I tested FT4 QSO mode import, but I didn't find any problems when FT4 is exported correctly - that means as Mode MFSK and submode FT4. Can you please give me the ADI record of the given QSO so I can check it?

ea5wa, 2023-12-22:
Here you have two adi files. In 0-sats.adi you can see some FT4 QSOs. As you said, FT4 is a submode, but why is FT8 a mode? In ultims.adi you can see some FT8 QSOs.

I've renamed the extension because adi is not supported by GitHub

[ultims.txt](https://github.com/foldynl/QLog/files/13756644/ultims.txt)
[0-sats.txt](https://github.com/foldynl/QLog/files/13756646/0-sats.txt)

ea5wa, 2023-12-22:
I've just realized that reports are not correct  
[image]

foldynl, 2023-12-23:
I verified provided files and I don't see any issue during import the file to QLog.

> As you said, FT4 is a submode, but why is FT8 a mode?

Good question, unfortunately, it is not a question to me. Please, see [ADIF specification](https://www.adif.org/314/ADIF_314.htm#Mode_Enumeration) where FT8 is defined as Mode and FT4 is defined as a submode of MFSK. 

> I've just realized that reports are not correct

It is difficult to implement this correctly. QLog currently defines a default RST per Mode. MFSK is a digital mode where the so-called RSQ (quality) report format can be used but in case of FT4 it is dB report format.

There is a good news that it can be easily changed.

1) FT4 QSOs are mostly received from external sources. So you don't have to worry about the default RST value because the external source provide a correct RST value.
2) If you want to enter FT4 QSO manually and the default RST value is unacceptable for you, it is possible to edit the default RST for MFSK in QLog's `Settings` -> `Modes` -> `Report`.

foldynl, 2024-01-05:
I am closing the issue. If you need more info, please, open a new issue report.

closed 2024-01-05 by foldynl