foldynl/QLog #300: No SunSDR support

RobbiePh, 2024-01-11:
I have tried several radio in rig control but none with display the proper modes?

DUSB on the rig is RTTY on the software.

 I am using the SunSDR2 .

Back to Log4OM, I think for the moment.

De Ei2iP

foldynl, 2024-01-12:
possible solutions are present in my comment [#299 ](https://github.com/foldynl/QLog/discussions/299#discussioncomment-8106046)

RobbiePh, 2024-01-15:
Found a work around, thanks. TCI Logger installed.

[image]

closed 2024-01-15 by RobbiePh

foldynl, 2024-01-16:
I'm glad it worked out. I would like to ask. I have no experience with SunSDR, but I looked at the TCI spec and it's not complicated. But it will be necessary to test it. Is there any SunSDR simulator software?