foldynl/QLog #301: clang linker failed

shtonich, 2024-01-13:
Hello.

On some Unix-like systems (not macx) the clang linker crashes.
<details>

<summary>Details</summary>

c++ -Wl,--undefined-version -fstack-protector-strong -pthread -Wl,-rpath,/usr/local/lib/qt5 -Wl,-rpath-link,/usr/local/lib/qt5 -o qlog AlertEvaluator.o  AppGuard.o  CWCatKey.o  CWDaemonKey.o  CWDummyKey.o  CWFldigiKey.o  CWKey.o  CWKeyer.o  CWWinKey.o  CallbookManager.o  Callsign.o  ClubLog.o  CredentialStore.o  DxServerString.o  Eqsl.o  Fldigi.o  GenericCallbook.o  Gridsquare.o  HRDLog.o  HamQTH.o  HostsPortString.o  KSTChat.o  LOVDownloader.o  LogLocale.o  LogParam.o  Lotw.o  MembershipQE.o  Migration.o  NetworkNotification.o  PropConditions.o  QRZ.o  QSLStorage.o  Rig.o  Rotator.o  SerialPort.o  Wsjtx.o  debug.o  main.o  zonedetect.o  AntProfile.o  BandPlan.o  CWKeyProfile.o  CWShortcutProfile.o  Data.o  Dxcc.o  MainLayoutProfile.o  RigProfile.o  RotProfile.o  RotUsrButtonsProfile.o  StationProfile.o  AdiFormat.o  AdxFormat.o  CSVFormat.o  JsonFormat.o  LogFormat.o  AlertTableModel.o  AwardsTableModel.o  DxccTableModel.o  LogbookModel.o  RigTypeModel.o  RotTypeModel.o  SqlListModel.o  WsjtxTableModel.o  AlertRuleDetail.o  AlertSettingDialog.o  AlertWidget.o  AwardsDialog.o  BandmapWidget.o  CWConsoleWidget.o  ChatWidget.o  ClockWidget.o  ClublogDialog.o  ColumnSettingDialog.o  DxFilterDialog.o  DxWidget.o  DxccTableWidget.o  EditLayoutDialog.o  EditLine.o  Eqsldialog.o  ExportDialog.o  HRDLogDialog.o  ImportDialog.o  InputPasswordDialog.o  KSTChatWidget.o  KSTHighlightRuleDetail.o  KSTHighlighterSettingDialog.o  LogbookWidget.o  LotwDialog.o  MainLayoutEditor.o  MainWindow.o  MapWebChannelHandler.o  MapWidget.o  NewContactWidget.o  OnlineMapWidget.o  PaperQSLDialog.o  QSLImportStatDialog.o  QSODetailDialog.o  QSOFilterDetail.o  QSOFilterDialog.o  QTableQSOView.o  QrzDialog.o  RigWidget.o  RotatorWidget.o  SettingsDialog.o  ShowUploadDialog.o  StatisticsWidget.o  SwitchButton.o  WebEnginePage.o  WsjtxFilterDialog.o  WsjtxWidget.o  qrc_i18n.o  qrc_flags.o  qrc_icons.o  qrc_res.o  moc_AlertEvaluator.o  moc_CWCatKey.o  moc_CWDaemonKey.o  moc_CWDummyKey.o  moc_CWKey.o  moc_CWKeyer.o  moc_CWWinKey.o  moc_CallbookManager.o  moc_Callsign.o  moc_ClubLog.o  moc_CredentialStore.o  moc_Eqsl.o  moc_Fldigi.o  moc_GenericCallbook.o  moc_HRDLog.o  moc_HamQTH.o  moc_HostsPortString.o  moc_KSTChat.o  moc_LOVDownloader.o  moc_LogParam.o  moc_Lotw.o  moc_MembershipQE.o  moc_Migration.o  moc_NetworkNotification.o  moc_PropConditions.o  moc_QRZ.o  moc_QSLStorage.o  moc_Rig.o  moc_Rotator.o  moc_SerialPort.o  moc_Wsjtx.o  moc_AntProfile.o  moc_CWKeyProfile.o  moc_CWShortcutProfile.o  moc_Data.o  moc_MainLayoutProfile.o  moc_RigProfile.o  moc_RotProfile.o  moc_RotUsrButtonsProfile.o  moc_StationProfile.o  moc_LogFormat.o  moc_AlertTableModel.o  moc_AwardsTableModel.o  moc_LogbookModel.o  moc_RigTypeModel.o  moc_RotTypeModel.o  moc_SqlListModel.o  moc_WsjtxTableModel.o  moc_AlertRuleDetail.o  moc_AlertSettingDialog.o  moc_AlertWidget.o  moc_AwardsDialog.o  moc_BandmapWidget.o  moc_CWConsoleWidget.o  moc_ChatWidget.o  moc_ClockWidget.o  moc_ClublogDialog.o  moc_ColumnSettingDialog.o  moc_DxFilterDialog.o  moc_DxWidget.o  moc_DxccTableWidget.o  moc_EditLayoutDialog.o  moc_EditLine.o  moc_Eqsldialog.o  moc_ExportDialog.o  moc_HRDLogDialog.o  moc_ImportDialog.o  moc_InputPasswordDialog.o  moc_KSTChatWidget.o  moc_KSTHighlightRuleDetail.o  moc_KSTHighlighterSettingDialog.o  moc_LogbookWidget.o  moc_LotwDialog.o  moc_MainLayoutEditor.o  moc_MainWindow.o  moc_MapWebChannelHandler.o  moc_MapWidget.o  moc_NewContactWidget.o  moc_OnlineMapWidget.o  moc_PaperQSLDialog.o  moc_QSLImportStatDialog.o  moc_QSODetailDialog.o  moc_QSOFilterDetail.o  moc_QSOFilterDialog.o  moc_QTableQSOView.o  moc_QrzDialog.o  moc_ShowUploadDialog.o  moc_StyleItemDelegate.o  moc_RigWidget.o  moc_RotatorWidget.o  moc_SettingsDialog.o  moc_StatisticsWidget.o  moc_SwitchButton.o  moc_WsjtxFilterDialog.o  moc_WsjtxWidget.o   -L/usr/local/lib -L -lhamlib -lqt5keychain /usr/local/lib/qt5/libQt5Charts.so /usr/local/lib/qt5/libQt5WebEngineWidgets.so /usr/local/lib/qt5/libQt5PrintSupport.so /usr/local/lib/qt5/libQt5QuickWidgets.so /usr/local/lib/qt5/libQt5Widgets.so /usr/local/lib/qt5/libQt5WebEngineCore.so /usr/local/lib/qt5/libQt5Quick.so /usr/local/lib/qt5/libQt5Gui.so /usr/local/lib/qt5/libQt5Sql.so /usr/local/lib/qt5/libQt5QmlModels.so /usr/local/lib/qt5/libQt5WebChannel.so /usr/local/lib/qt5/libQt5Qml.so /usr/local/lib/qt5/libQt5Network.so /usr/local/lib/qt5/libQt5Xml.so /usr/local/lib/qt5/libQt5Positioning.so /usr/local/lib/qt5/libQt5SerialPort.so /usr/local/lib/qt5/libQt5DBus.so /usr/local/lib/qt5/libQt5Core.so -lGL
ld: error: undefined symbol: rig_get_ptt
>>> referenced by Rig.cpp:182 (core/Rig.cpp:182)
>>>               Rig.o:(Rig::update())

ld: error: undefined symbol: rig_get_freq
>>> referenced by Rig.cpp:223 (core/Rig.cpp:223)
>>>               Rig.o:(Rig::update())

ld: error: undefined symbol: rig_get_mode
>>> referenced by Rig.cpp:269 (core/Rig.cpp:269)
>>>               Rig.o:(Rig::update())

ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
c++: error: linker command failed with exit code 1 (use -v to see invocation)
*** [qlog] Error code 1

</details>

The reason for the error is an extra option **-L** before -lhamlib.

I'm not a qmake expert, but the LIBS variable is alarming.

My workaround

```
--- QLog.pro.orig       2024-01-08 21:11:08 UTC
+++ QLog.pro
@@ -391,7 +391,7 @@ unix:!macx {
    INSTALLS += target desktop icon metainfo
 
    INCLUDEPATH += /usr/local/include
-   LIBS += -L/usr/local/lib -lhamlib
+   LIBS = -L/usr/local/lib -lhamlib
    equals(QT_MAJOR_VERSION, 6): LIBS += -lqt6keychain
    equals(QT_MAJOR_VERSION, 5): LIBS += -lqt5keychain
 }
```

foldynl, 2024-01-13:
Please, could you specify OS version, clang version, qt version

shtonich, 2024-01-14:
> Please, could you specify OS version, clang version, qt version

```
> freebsd-version -kru
13.2-RELEASE-p8
13.2-RELEASE-p8
13.2-RELEASE-p9
```

```
> c++ --version
FreeBSD clang version 14.0.5 (https://github.com/llvm/llvm-project.git llvmorg-14.0.5-0-gc12386ae247c)
Target: x86_64-unknown-freebsd13.2
Thread model: posix
InstalledDir: /usr/bin
```
```
> qmake-qt5 --version
QMake version 3.1
Using Qt version 5.15.10 in /usr/local/lib/qt5
```

foldynl, 2024-01-14:
Unfortunately, your patch breaks the build process where external variables are defined via qmake parameters. I went through your compilation report, and I can see something different than what I have with me. Your compilation contains an empty `-L` between `-L/usr/local/lib` and `-lhamlib`. This doesn't happen to me, and I don't know why. My compilation contains the same sequence of `-L` and `-l` as is present in the .pro file.

It needs to be investigated, and I have to understand it before any patch is made. I'll try to suggest something, and then I'll let you know.

foldynl, 2024-01-15:
Please, could you test this patch?

```
diff --git a/QLog.pro b/QLog.pro
index f8ba6a2..6dbc681 100644
--- a/QLog.pro
+++ b/QLog.pro
@@ -369,8 +369,29 @@ isEmpty(HAMLIBVERSION_MAJOR) {
    }
 }
 
-INCLUDEPATH += $$HAMLIBINCLUDEPATH $$QTKEYCHAININCLUDEPATH $$PTHREADINCLUDEPATH
-LIBS += -L$$HAMLIBLIBPATH -L$$QTKEYCHAINLIBPATH -L$$PTHREADINCLUDEPATH
+!isEmpty(HAMLIBINCLUDEPATH) {
+   INCLUDEPATH += $$HAMLIBINCLUDEPATH
+}
+
+!isEmpty(QTKEYCHAININCLUDEPATH) {
+   INCLUDEPATH += $$QTKEYCHAININCLUDEPATH
+}
+
+!isEmpty(PTHREADINCLUDEPATH) {
+   INCLUDEPATH += $$PTHREADINCLUDEPATH
+}
+
+!isEmpty(HAMLIBLIBPATH) {
+   LIBS += -L$$HAMLIBLIBPATH
+}
+
+!isEmpty(QTKEYCHAINLIBPATH) {
+   LIBS += -L$$QTKEYCHAINLIBPATH
+}
+
+!isEmpty(PTHREADINCLUDEPATH) {
+   LIBS += -L$$PTHREADINCLUDEPATH
+}
 
 unix:!macx {
    isEmpty(PREFIX) {

```

foldynl, 2024-01-15:
Many thanks for your confirmation. The patch will be a part of the next release.

commit a2ae1ff8 (2024-02-10): Fixed #301 - clang linker failed
closed 2024-02-10 by foldynl