foldynl/QLog #305: cw daemon configuration help

ik3ocd, 2024-01-19:
good morning from flavio ik3ocd, sorry for my return here in the forum, I'm struggling with cwdaemon I configured everything as per your help, what doesn't happen is no action in transmission on my yaesu ftdx 3000 I'll start by saying that the virtual port /dev/ ttyUSB0 works with rig, now I have a free serial port /dev/ttyUSB1 for possible ptt, and this is where it is not clear to me how to configure cwdaemon to show the serial and baud rate to send transmission via the yaesu ftdx 3000 macros thanks for your help. 73 de ik3ocd flavio da venice.
[image]
[image]

foldynl, 2024-01-19:
There is no need to enter any Serial Port parameters for cwdaemon in QLog. CWDaemon is a background running process that should be started before requesting a connect key in QLog. It can be started manually or automatically (more `man cwdaemon`). Only the CWDaemon parameters determine which port will be used for PTT etc. It's good to run `cwdaemon --help`

Example:
If I want to start cwdatemon so that it only produces beeper via sound card then you I can call :

`cwdaemon -x pulse`

then I run QLog. The settings are the same as what you attached in the pictures.

If I want to use ttyUSB1 then it is necessary to start cwdaemon like this

`cwdaemon -d /dev/ttyUSB1`

cwdaemon has many parameters, I recommend to read `man cwdaemon` for better understanding all parameters and use parameter '-n' for debugging your cwdaemon parameters.

ik3ocd, 2024-01-19:
hello and good evening, thanks for the answer, it was what was confirmed to me. 73 Best regards and thanks for your excellent work on qlog. flavio ik3ocd

closed 2024-01-19 by foldynl