foldynl/QLog #306: Problem importing QSOs

ea5wa, 2024-01-21:
I have this error message when trying to import some QSOs

[QSO#1]: Error - Cannot insert to database - FOREIGN KEY constraint failed No es posible obtener la fila (21/01/2024; DP0POL; MFSK)
[QSO#2]: Error - Cannot insert to database - FOREIGN KEY constraint failed No es posible obtener la fila (21/01/2024; PY2PIM; MFSK)
[QSO#3]: Error - Cannot insert to database - FOREIGN KEY constraint failed No es posible obtener la fila (21/01/2024; EA4DFE; SSB)

Same adif file is correctly open with AdifMaster and uploaded to qrz.com without any error

foldynl, 2024-01-21:
Please, could you send me the ADIF records for mentioned QSOs?

ea5wa, 2024-01-21:
I tried to attached it here but is not allowed

foldynl, 2024-01-22:
thanks for providing the ADIF file via email.

Unfortunately, your ADIF file contains many invalid fields. Some of them are critical, some are not critical for import.

Critical:
- `SWL` is defined as [boolean](https://adif.org/314/ADIF_314.htm#Boolean) in ADIF spec - should be `Y`, `y`, `N` or `n`. Your file contains `0`
- `QSO_RANDOM` is define also as boolean, your file contains `0`

there are many non-critical errors that do not meet the ADIF specification. Especially fields with `-1` like `MY_FISTS`, `K_INDEX`, `ITU_ZONE`, `MAX_BURSTS`, `SFI`, `TEN-TEN`, `UKSMG`. These fields must contain the positive integer value.

the last thing is the incomplete definition of mode. It does not beak ADIF, but it's quite unusual to define only the `Mode` and not the `submode`, especially in the case of MFSK, where there are several submodes - you probably should have FT4 submode there.

If you want to import the ADIF file to QLog then the minimum modification is  to change the field `SWL` and `QSO_RANDOM` from `0` to `N`. All non-critical error are accepted by QLog. Unfortunately, due to considerable freedom on the QLog import, these negative fields are saved with an incorrect value, so exporting these fields from Qlog will cause ADIF to violate some definitions again.

ea5wa, 2024-01-22:
Hi Ladislav:

Thanks for the explanation

Same adif file is imported to other platforms (software) without any problem. Why?

73 de EA5WA Juan Carlos

ea5wa, 2024-01-22:
> it's a good question in the wrong place. It is necessary to ask the
> platform administrators.
>
> ADIF clearly specifies the fields types. The question is which fields
> other platforms check and if the platforms use them internally.. For
> example, Clublog imports only a few selected fields and ignores others.
> I can't say what and how other platforms check imported fields.
>
> QLog checks all enum fields.

Many thanks Ladislav

⁣73 de EA5WA Juan Carlos

closed 2024-01-22 by foldynl