foldynl/QLog #307: a desire for functionality

ik3ocd, 2024-01-25:
good morning sir ladislav, best wishes for the future in the function key cluster section with fast bandwidth filtering, QSO label printing management with glabels for linux. thanks for listening, at the moment I don't see any functioning problems on qlog using fedora linux 39. hello and best regards e 73 from ik3ocd flavio

foldynl, 2024-01-25:
Glabel is already supported . If you have a Glabel's template with your QSL card then GLabel can import CSV file with QSO data. QLog can export QSOs to CSV file therefore I do not see any problem to print QSL via Glabel.

> best wishes for the future in the function key cluster section with fast bandwidth filtering,

Could you please provide a more detailed description of what this function should do?

ik3ocd, 2024-01-25:
Hi Ladislav good evening, yes thanks for the reply, in reality I had already worked with CVS glabels but with little success I have to put more effort into it on my part, for the cluster function keys example button 160 80 40 30 20 15 12 10 6 2 8 (the bands most used frequency) I mean the quick possibility of having spot filter buttons for each band and that is filtering the spots that pass for example on 40 meters, without going to sub qlog configurations on the cluster, sorry for my not perfect English. thank you for your patience. 73 of Flavian ik3ocd

closed 2024-01-25 by ik3ocd