foldynl/QLog #311: database award question

ik3ocd, 2024-01-29:
good morning ladislav, a question:
how dxcc, wpx, iota, sota pota  etc databases are updated. Is there a possibility to update manually or on demand via qlog or is there a future implementation of things to do? will there also be a statistic for award sota pota  etc..?.
thanks for the reply and have a nice day. 73 of Flavio ik3ocd

foldynl, 2024-01-29:
DXCC, IOTA SOTA POTA, WWFF, SATLIST, Club Membership records are regularly updated by QLog. Qlog finds out how old the databases are and downloads them from the Internet if it is needed. It can't be done manually

There is no database for WPX. WPX is derived from callsign and is calculated directly when QSO is saved.

DXCC IOTA WPX Statistics for your station are then calculated directly when the given dialog window is displayed

> will there also be a statistic for award sota pota etc..?.

not in the near future. I don't mind if someone sends it as a Pull Request.

foldynl, 2024-08-29:
SOTA, POTA, WWFF statistics are part of v0.38

closed 2024-08-29 by foldynl

ik3ocd, 2024-08-30:
Great! Grazie ladislav per questa grande gestione 🙌  pota,sota,wwff.  73
de ik3ocd Flavio