foldynl/QLog #314: Networking and MSHV

ra3sad, 2024-02-01:
Database fields **Start_time** and **End_time** is empty when receiving QSO (**Logged** packet) from MSHV software via UDP. Validation fails.

foldynl, 2024-02-01:
Please, could you share MSHV version?

ra3sad, 2024-02-01:
MSHV ver 2.64

ra3sad, 2024-02-01:
Ok, I will update and test the latest version of MSHV.
No, I only use Linux, that's why your software is very interesting for me.
Tnx!

foldynl, 2024-02-01:
I have performed some tests. I used MSHV 2.73 and QLog 0.31 (QT 5.x) . Both were running on Ubuntu 22.06. MSHV installed from deb package and I found no issue. Everything looks good. 

The only thing that doesn't seem to me is that MSHV did not send a notification about the mode change, but that may be connected with the quick setting of my MSHV. Please test the latest version and let me know.

ra3sad, 2024-02-02:
Tested MSHV version 2.73 and QLog from latest git (Qt 5.15.2). Everything is fine. Thanks!

foldynl, 2024-02-02:
Perfect...good news...

closed 2024-02-02 by foldynl