foldynl/QLog #316: ADIF export issue

ra3sad, 2024-02-11:
Hi!
There are problems with export in ADIF format - NAME and QTH fields are missing (but in table is not empty and looks fine), but if they exist - their contents are - `<qth:6>. ,  -` . This applies only to the Cyrillic characters
System charset is UTF-8. Debian 12. Software version - 0.31.0
Thanks, Alex.

foldynl, 2024-02-11:
Alex, ADIF format has two formats - ADI and ADX. ADI is defined as pure ASCII format what means that there has not to be any non-ASCII character. On the other hand, ADX support UTF-8 charset. 

Because many applications do not comply with this, QLog is able to import UTF-8 fields even in the case of ADI. On the other hand, QLog never exports non-ASCII char to ADI. QLog tries to do some conversion, but in the ADI field, the non-ASCII char is always overwritten, as in your case.

If you want to export non-ASCII chars, then you need to use a more suitable format like ADX or CSV.

ra3sad, 2024-02-12:
Thanks, I understand the logic. I solved the problem for myself by editing the source code. Maybe add a checkbox to export dialog to select whether to filter non-ASCII characters or not?

foldynl, 2024-02-13:
checkbox is not a good idea, because it would totally disturb the consistency of the data in ADI. The only thing I could accept is a warning that such a conversion occurs.

commit 65fc34bb (2024-04-25): Fixed #350 - removeAccents: New algorithm that uses the Perl library Text-Unidecode/iconv approach

The algorithm is also two times faster than the previous one.
This commit also fix an issue:

Fixed #316 - New Translit approach guarantees that Cyrillic characters will be transcribed into ASCII
closed 2024-05-03 by foldynl