foldynl/QLog #319: new database

ik3ocd, 2024-02-16:
good morning, will it be possible in your future projects on qlog to create separate databases, perhaps there is already this possibility? but I don't see it. Thank you for your always kind response. 73 de ik3ocd flavio .

foldynl, 2024-02-17:
The original QLog, before my changes, was designed as a single-database log, and I do not intend to alter this original design. With the help of filters, you can select the QSOs that interest you. In the multi-database concept, I see inconvenience for the operator in searching for a specific QSO rather than a benefit.

It is true that I have encountered a problem several times that could easily be solved with the help of multiple databases, but officially it still remains a single-database log.

To be honest, there is an option to have several different QLog instances with different settings. However, this option is intended only for testing purposes, not for regular usage.

As for future plans, I receive several requests per week, but unfortunately, I am still a single developer. So, I will answer truthfully: I don't plan to implement this in the near future.

Could you please describe why separate databases are useful for you?

Please, could you describe why separate database are useful for you?

ik3ocd, 2024-02-17:
Hi, thank you for replying and your further consideration regarding this on database, my request is simple, I do various activities on the laptop sometimes I find it convenient to observe my connections made in various locations for greater management regarding e-QSL by confirming separately for each /p activity, also have a database distinction between hf and vhf/uhf connections. Anyway, thanks for your excellent work and future improvements. Best regards

foldynl, 2024-02-17:
But it's a shame to have information across multiple databases and not utilize the data obtained from previous QSOs. You can create filters for all the information logged in QLog to simulate separate database. Sending a QSO, depending on the locator, can be achieved via QLog eQSL  Export - it's necessary to change the QTH Nickname before sending.

It's a shame to lose all information due to separate database, don't you think?

ik3ocd, 2024-02-17:
Yes, I think you are right and I gladly accept your suggestion and by working on the filters you can achieve the same result without having various databases.

closed 2024-03-17 by foldynl