foldynl/QLog #325: Hamlib integration not working in v0.33?

g0fcu, 2024-03-09:
Following installation of v0.33 (Xubuntu 22.04) the connection to the rig via rigctld is not working. I start rigctld from a terminal as I have done previously and set up the rig in qlog like this:
[image]

When trying Connect to Rig qlog hangs and has to be killed. If rigctld is not running qlog reports correctly that it cannot connect to the rig. If I set the rig up as a serial connection via USB the same behaviour occurs.

I am hoping this is an issue I have caused but have not changed anything. The only thing I did that may have caused a problem was select TCI as the interface before correctly selecting Hamlib.

If I run rigctl in non-daemon mode I get a response from the rig, e.g. when I enter the 'f' command the correct frequency the rig is tuned to is returned.

73 Simon
G0FCU.

rsaue, 2024-03-09:
I have the same overall experience as Simon using the precompiled install package on a windows 11 system, and on ubuntu linux using self-compiled binary from github code. This also affects TCI on windows (have not tried TCI on linux). 

Version 0.32 worked well on both systems. 

73 de LB2EG Richard

foldynl, 2024-03-09:
Unfortunately, I can confirm there is an issue. I will try to find it and release a hot fix ASAP

foldynl, 2024-03-09:
Please, would it be possible to test following [qlog.exe](https://github.com/foldynl/QLog/releases/download/v0.33.0/qlog.exe) file. Just replace the old qlog.exe file with the new one and test it. It should fix the issue. 

if your test is successful, I will release a complete new release.

rsaue, 2024-03-09:
Hi, the new exe works ok with Hamlib CAT connection to my FTDX-10, and on TCP/IP to Thetis (Anan-200D). The TCI connection works OK to my SunSD2-DX (using EESDR v 3 latest public beta). 

I have problems connecting Qlog to Thetis using TCI. The connection disconnects after a few seconds. We never tested the TCI with Thetis, only with SunSDR, so that may be something to check out later. 

73 de LB2EG Richard

foldynl, 2024-03-09:
Richard, many thanks for your test.  There were several errors, but mainly I made a mistake during testing.

I am going to make a release.

I know about the Thetis issue, but I don't know what it is. At first I thought it was because I didn't have the right rig connected, but you confirmed that it doesn't work either. That's why I had to create a TCI simulator so that I could at least test something. 

It is interesting for me to know that wfview has experimental support for TCI as you mentioned on Mastodon. That would be good news for me. If it is possible, please, send me an link where the experimental support is implemented. 

Many thanks for you assistance

rsaue, 2024-03-09:
[https://gitlab.com/eliggett/wfview/-/tree/creator-widgets?ref_type=heads](url)

closed 2024-03-09 by foldynl