foldynl/QLog #327: TCI compatibility ??

EA5SW, 2024-03-12:
Hello, the TCI cat control  connect and  disconnect in a fraction of second when connected to Thetis for Hermes Lite.

I tried some ports another that usual (40001,30001) and no luck.

Thetis works with LOG4OM and the utils of Expert SDR software (Radiosync,RadioSpot9 etc...

Thanks in advance..

rsaue, 2024-03-12:
The QLOG TCI implementation was only tested in the field with SunSDR EESDR v.3. It is known that it does not work with the TCI server in Thetis, and I believe Ladislav will look further into this, maybe with field testing assistance from those among us who uses a radio powered with Thetis. 

73 de LB2EG Richard

foldynl, 2024-03-12:
As Richard mentioned. Unfortunately, I have no rig which is compatible with Thetis. Please, could you send me Thetis TCI log with your attempts to connect? You can find the log in Setup->Serial/Network/Midi CAT tab->Network Tab-> TCI Server press Show Log.
 
In my case - no rig connected, Thetis sends STOP command right after the opening command sequence, which is incomprehensible to me at this moment. QLog currently stops the connection when this command is received. Maybe, it should be ignored but when I modified the Qlog to ignore the command, Thetis repeats STOP command again.

EA5SW, 2024-03-12:
I have both, Expert3 and Hermes, now the  Expert 3 are disconnected, but i connect tomorrow to test.

The log are here:

> stop;
< spot:R5KH, ,3573000,4294967167, ;
< TRX:0;
< VFO:0,0;
> ready;
> start;
> mon_volume:-30.0;
> mon_enable:false;
> rx_mute:1,false;
> rx_mute:0,false;
> mute:false;
> tune:1,false;
> tune:0,false;
> trx:1,false;
> trx:0,false;
> tx_enable:1,false;
> tx_enable:0,true;
> split_enable:1,false;
> split_enable:0,true;
> rx_enable:1,false;
> rx_enable:0,true;
> modulation:1,LSB;
> modulation:0,USB;
> vfo:1,1,18073050;
> vfo:1,0,18073050;
> vfo:0,1,18073050;
> vfo:0,0,18140000;
> if:1,1,0;
> if:1,1,0;
> if:0,1,0;
> if:0,0,0;
> dds:1,18073050;
> dds:0,18140000;
> modulations_list:AM,SAM,DSB,LSB,USB,NFM,FM,DIGL,DIGU,CWL, CWU, CW;
> if_limits:-96000,96000;
> vfo_limits:0,38400000;
> channels_count:2;
> trx_count:2;
> receive_only:false;
> device:SunSDR2PRO;
> protocol:ExpertSDR3,1.8;

EA5SW, 2024-03-12:
This is when I start LOG4OM

> vfo:0,1,18073050;
> split_enable:0,false;
< SPLIT_ENABLE:0;
< VFO:0,1;
< VFO:0,0;
> split_enable:0,false;
< SPLIT_ENABLE:0;
< VFO:0,1;
< VFO:0,0;
> modulation:0,USB;
< MODULATION:0;
> vfo:0,0,18123000;
> split_enable:0,false;
< SPLIT_ENABLE:0;
< VFO:0,1;
< VFO:0,0;
> ready;
> start;
> mon_volume:-30.0;
> mon_enable:false;
> rx_mute:1,false;
> rx_mute:0,false;
> mute:false;
> tune:1,false;
> tune:0,false;
> trx:1,false;
> trx:0,false;
> tx_enable:1,false;
> tx_enable:0,true;
> split_enable:1,false;
> split_enable:0,false;
> rx_enable:1,false;
> rx_enable:0,true;
> modulation:1,LSB;
> modulation:0,USB;
> vfo:1,1,18073050;
> vfo:1,0,18073050;
> vfo:0,1,18073050;
> vfo:0,0,18123000;
> if:1,1,0;
> if:1,1,0;
> if:0,1,0;
> if:0,0,0;
> dds:1,18073050;
> dds:0,18123000;
> modulations_list:AM,SAM,DSB,LSB,USB,NFM,FM,DIGL,DIGU,CWL, CWU, CW;
> if_limits:-96000,96000;
> vfo_limits:0,38400000;
> channels_count:2;
> trx_count:2;
> receive_only:false;
> device:SunSDR2PRO;
> protocol:ExpertSDR3,1.8;

EA5SW, 2024-03-12:
TCI LOG for SunSDR ExpertSDR3

23-17-59:216 client: spot:C31VM, ,7075900,4283949961, ;
23-18-08:793 client: spot:JF2TJJ, ,21075600,4283949961, ;
23-18-09:435 server: app_focus:true;
23-18-09:435 server: app_focus:true;
23-18-09:761 server: start;
23-18-09:761 server: start;
23-18-09:904 server: tx_frequency:7085000;
23-18-09:904 server: tx_frequency:7085000;
23-18-12:462 server: app_focus:false;
23-18-12:462 server: app_focus:false;
23-18-16:955 server: protocol:ExpertSDR3,1.10;
23-18-16:955 server: device:SunSDR2PRO;
23-18-16:955 server: receive_only:false;
23-18-16:955 server: trx_count:2;
23-18-16:955 server: channels_count:2;
23-18-16:955 server: vfo_limits:0,160000000;
23-18-16:955 server: if_limits:-156250,156250;
23-18-16:955 server: modulations_list:AM,LSB,USB,CW,NFM,WSPR,FT8,FT4,JT65,JT9,RTTY,BPSK,DIGL,DIGU,WFM,DRM;
23-18-16:955 server: iq_samplerate:48000;
23-18-16:955 server: audio_samplerate:48000;
23-18-16:955 server: volume:-4;
23-18-16:955 server: mute:true;
23-18-16:955 server: mon_volume:-22;
23-18-16:955 server: mon_enable:false;
23-18-16:955 server: digl_offset:0;
23-18-16:955 server: digu_offset:0;
23-18-16:955 server: dds:0,7044100;
23-18-16:955 server: dds:1,7100000;
23-18-16:955 server: if:0,0,40900;
23-18-16:955 server: if:0,1,22900;
23-18-16:955 server: if:1,0,0;
23-18-16:955 server: if:1,1,0;
23-18-16:955 server: vfo:0,0,7085000;
23-18-16:955 server: vfo:0,1,7067000;
23-18-16:955 server: vfo:1,0,7100000;
23-18-16:955 server: vfo:1,1,7100000;
23-18-16:955 server: modulation:0,LSB;
23-18-16:955 server: modulation:1,AM;
23-18-16:955 server: agc_mode:0,normal;
23-18-16:955 server: agc_gain:0,80;
23-18-16:955 server: agc_mode:1,normal;
23-18-16:955 server: agc_gain:1,100;
23-18-16:955 server: rx_filter_band:0,-3000,-40;
23-18-16:955 server: rx_filter_band:1,-8000,8000;
23-18-16:955 server: rx_nb_enable:0,false;
23-18-16:955 server: rx_nb_param:0,70,20;
23-18-16:955 server: rx_nb_enable:1,false;
23-18-16:955 server: rx_nb_param:1,70,20;
23-18-16:955 server: rx_bin_enable:0,false;
23-18-16:955 server: rx_bin_enable:1,false;
23-18-16:955 server: rx_nr_enable:0,false;
23-18-16:955 server: rx_nr_enable:1,false;
23-18-16:955 server: rx_anc_enable:0,false;
23-18-16:955 server: rx_anc_enable:1,false;
23-18-16:955 server: rx_anf_enable:0,false;
23-18-16:955 server: rx_anf_enable:1,false;
23-18-16:955 server: rx_apf_enable:0,false;
23-18-16:955 server: rx_apf_enable:1,false;
23-18-16:955 server: rx_dse_enable:0,false;
23-18-16:955 server: rx_dse_enable:1,false;
23-18-16:955 server: rx_nf_enable:0,false;
23-18-16:955 server: rx_nf_enable:1,false;
23-18-16:955 server: rx_enable:1,false;
23-18-16:955 server: tx_enable:0,true;
23-18-16:955 server: tx_enable:1,true;
23-18-16:955 server: lock:0,false;
23-18-16:955 server: lock:1,false;
23-18-16:955 server: rx_channel_enable:0,1,false;
23-18-16:955 server: rit_enable:0,false;
23-18-16:955 server: xit_enable:0,false;
23-18-16:955 server: split_enable:0,false;
23-18-16:955 server: rit_offset:0,0;
23-18-16:955 server: xit_offset:0,0;
23-18-16:955 server: sql_enable:0,false;
23-18-16:955 server: sql_level:0,-80;
23-18-16:955 server: rx_mute:0,false;
23-18-16:955 server: rx_volume:0,0,0;
23-18-16:955 server: rx_balance:0,0,0;
23-18-16:955 server: rx_volume:0,1,0;
23-18-16:955 server: rx_balance:0,1,0;
23-18-16:955 server: rx_channel_enable:1,1,false;
23-18-16:955 server: rit_enable:1,false;
23-18-16:955 server: xit_enable:1,false;
23-18-16:955 server: split_enable:1,false;
23-18-16:955 server: rit_offset:1,0;
23-18-16:955 server: xit_offset:1,0;
23-18-16:955 server: sql_enable:1,false;
23-18-16:955 server: sql_level:1,-80;
23-18-16:955 server: rx_mute:1,false;
23-18-16:955 server: rx_volume:1,0,0;
23-18-16:955 server: rx_balance:1,0,0;
23-18-16:955 server: rx_volume:1,1,0;
23-18-16:955 server: rx_balance:1,1,0;
23-18-16:955 server: cw_macros_speed:29;
23-18-16:955 server: cw_macros_delay:10;
23-18-16:955 server: cw_keyer_speed:32;
23-18-16:955 server: drive:0,0;
23-18-16:955 server: tune_drive:0,12;
23-18-16:955 server: drive:1,0;
23-18-16:955 server: tune_drive:1,0;
23-18-16:955 server: trx:0,false;
23-18-16:955 server: tune:0,false;
23-18-16:955 server: iq_stop:0;
23-18-16:955 server: trx:1,false;
23-18-16:955 server: tune:1,false;
23-18-16:955 server: iq_stop:1;
23-18-16:955 server: app_focus:false;
23-18-16:955 server: tx_frequency:7085000;
23-18-16:955 server: audio_stream_sample_type:float32;
23-18-16:955 server: audio_stream_channels:2;
23-18-16:955 server: start;
23-18-16:964 server: vfo_lock:0,0,false;
23-18-16:964 server: vfo_lock:0,1,false;
23-18-16:964 server: vfo_lock:1,0,false;
23-18-16:964 server: vfo_lock:1,1,false;
23-18-17:198 server: ready;
23-18-17:199 client: VFO:0,0;
23-18-17:199 server: vfo:0,0,7085000;
23-18-17:199 client: TRX:0;
23-18-17:199 server: trx:0,false;
23-18-17:199 client: MODULATION:0;
23-18-17:199 server: modulation:0,LSB;
23-18-17:199 client: DRIVE:0;
23-18-17:199 server: drive:0,0;
23-18-17:199 client: RIT_OFFSET:0;
23-18-17:199 server: rit_offset:0,0.0;
23-18-17:199 client: RIT_ENABLE:0;
23-18-17:199 server: rit_enable:0,false;
23-18-17:199 client: XIT_OFFSET:0;
23-18-17:199 server: xit_offset:0,0.0;
23-18-17:199 client: XIT_ENABLE:0;
23-18-17:199 server: xit_enable:0,false;
23-18-17:199 client: CW_MACROS_SPEED;
23-18-32:017 client: spot:IR0IDP, ,5353000,4283949961, ;

foldynl, 2024-03-13:
Many thanks for the log. They are very useful. Just to be sure. the log from "TCI LOG for SunSDR ExpertSDR3" test. Is it SunSDR ExpertSDR3 and QLog? Do you see the same issue - disconnect - in this case?

EA5SW, 2024-03-13:
Next step: 

> Many thanks for the log. They are very useful. Just to be sure. the log from "TCI LOG for SunSDR ExpertSDR3" test. Is it SunSDR ExpertSDR3 and QLog? Do you see the same issue - disconnect - in this case?

Yes is my SundSDR pro ExpertSDR3 with QLOG.

All of this in Windows11, now im compiling in Linux Debian and do some more tests.

EA5SW, 2024-03-13:
I compile in Linux and ignore the stop command in Qlog, for QLOG connection are OK, Thetis send a stop.
The Rig model 0 in QLOG are VFO A  and  Rig model 1 is VFO B

The Rig window update Freq & mode any time you connect, but Thetis always send a stop... it's possilble need a short timing ???

foldynl, 2024-03-13:
I can see couple of issue in QLog's TCI:

1) missing disconnect event handler - it is not related to the original issue but it has to be added.
2) `client: spot:IR0IDP, ,5353000,4283949961, ;` - Mode is missing here. Currently I don't know why because it was tested locally without any issue but your log shows that there is no mode in the spot command.. this could potentially affect Thetis as it does not meet the TCI specification. And `stop` sending them right after this command. Maybe, please, could you disable Spot2Rig function and test it again against Thetis? If there is the same result - disconnect. 
3) Timing - it is possible. QLog's TCI driver does not add additional sleep as is the case with other QLog's drivers (Hamlib, Omnirig etc). Maybe it is necessary to really give it some time. But I would be surprised if it is necessary, because TCI protocol is designed as a multi-user protocol, where a request can be received at any time.

I'll try to check also the Thetis sources if I can't find anything there.

EA5SW, 2024-03-13:
I compile again with no TRX command and works OK, no more disconnections

foldynl, 2024-03-13:
What Thetis version are you using?

foldynl, 2024-03-13:
and isn't it just that the wrong SPOT command without the Mode field (field after callsign) wasn't sent to Thetis? I can see that TRX command is already implemented in your version of Thetis.

EA5SW, 2024-03-13:
> and isn't it just that the wrong SPOT command without the Mode field (field after callsign) wasn't sent to Thetis? I can see that TRX command is already implemented in your version of Thetis.

No, the spot command without mode that come from Radiospot dont change the mode, but are display in freq and dont make a error.

The spot from QLog works perfect, change freq and mode.

Both are displayed at same time in Thetis with different colours.

The only problem for now is the TRX command  for my installation at least, cat works perfect in both directions,in & out.

foldynl, 2024-03-13:
It does not make sense to me.

Based on Thetis code:

1) it cannot be a problem with the SPOT command, because in the case of a missing MODE, it is automatically filled in somehow - what you confirmed above.
2) TRX - it's a mystery to me. Command is supported by Thetis. I don't think it can cause a STOP command
3) The STOP command is also sent if the rig is not connected but I assume that in case of a test, you had connected and switched on your rig.

EA5SW, 2024-03-13:
I return the TCIDrv.cpp and TCIDrv.h at the default and made some more tests.

foldynl, 2024-03-13:
>> The STOP command is also sent if the rig is not connected but I assume that in case of a test, you had connected and switched on your rig.

Sorry, you had to connected the rig because the stop command because the command would have to be sent in the initialization phase of connection. That's how I see it, who doesn't have a real rig connected.

rsaue, 2024-03-13:
I see the same in the latest Thetis for ANAN radios. QLOG only sends the VFO and the TRX command, and Thetis replies with the STOP right after having received the TRX command from QLOG. 

> stop;
< TRX:0;
< VFO:0,0;

Log4om does not send the TRX-command on startup. LOGHX3 sends neither VFO nor TRX command upon connection. 

73 de LB2EG Richard

EA5SW, 2024-03-13:
I put // in front of send TRX lines in TCIDrv.cpp and works perfect, appears that the TRX command with Thetis are buggy or not supported.

foldynl, 2024-03-13:
And you are able to control PTT via QLog using `ALT + \`. This shortcut sends TRX command, but with different parameters.

EA5SW, 2024-03-13:
> And you are able to control PTT via QLog using `ALT + \`. This shortcut sends TRX command, but with different parameters.

In my Debian this shortcut don't work...  need to change to another key

rsaue, 2024-03-13:
I know. But I have not the system set up for compiling QLog on windows (I compile Thetis from github with WS2022, but have not walked the last mile setting up Qt compiling environment here). Did now compile qlog on a rpi5 in shack, to test, but have to figure out the firewall to access the ports on Thetis on win11...

foldynl, 2024-03-13:
Unfortunately, it is not possible to change the shortcut in Qlog now.

thank you both for your assistance. I think that at the moment I am able to prepare a "workaround" for TCI/Thetis, which does not have any significant effect on the SunSDR ExpertSDR3. I will remove the TRX in the initialization phase. It would be good to confirm if it is possible to send a TRX command to control the PTT (pressing `ALT + \`).

EA5SW, 2024-03-13:
It's possible assign the PTT to F12 for example ???

foldynl, 2024-03-13:
You can change it in source code

```
diff --git a/ui/MainWindow.cpp b/ui/MainWindow.cpp
index 7baafd4..fa73bbf 100644
--- a/ui/MainWindow.cpp
+++ b/ui/MainWindow.cpp
@@ -245,7 +245,7 @@ MainWindow::MainWindow(QWidget* parent) :
     /*************/
     /* SHORTCUTs */
     /*************/
-    QShortcut *shortcut = new QShortcut(QKeySequence(Qt::ALT | Qt::Key_Backslash),
+    QShortcut *shortcut = new QShortcut(QKeySequence(Qt::Key_F12),
                                         this,
                                         SLOT(shortcutALTBackslash()),
                                         nullptr, Qt::ApplicationShortcut);
```

EA5SW, 2024-03-13:
> You can change it in source code
>
> ```
> diff --git a/ui/MainWindow.cpp b/ui/MainWindow.cpp
> index 7baafd4..fa73bbf 100644
> --- a/ui/MainWindow.cpp
> +++ b/ui/MainWindow.cpp
> @@ -245,7 +245,7 @@ [...]

This works OK, PTT active, only TX don't switch TX<>RX

EA5SW, 2024-03-13:
> app_focus:false;
> app_focus:false;
> trx:0,false;
> trx:0,false;
> app_focus:true;
> app_focus:true;
< TRX:0,true;
> trx:0,true;
> trx:0,true;
< TRX:0,true;
< CW_MACROS_SPEED;
< XIT_ENABLE:0;
< XIT_OFFSET:0;
< RIT_ENABLE:0;
< RIT_OFFSET:0;
> drive:0,50;
< DRIVE:0;
> modulation:0,CW;
> vfo:0,0,14038200;
< MODULATION:0;
< VFO:0,0;
> ready;
> start;
> mon_volume:-30.0;
> mon_enable:false;
> rx_mute:1,false;
> rx_mute:0,false;
> mute:false;
> tune:1,false;
> tune:0,false;
> trx:1,false;
> trx:0,false;
> tx_enable:1,false;
> tx_enable:0,true;
> split_enable:1,false;
> split_enable:0,false;
> rx_enable:1,false;
> rx_enable:0,true;
> modulation:1,LSB;
> modulation:0,CW;
> vfo:1,1,28054200;
> vfo:1,0,28054200;
> vfo:0,1,28054200;
> vfo:0,0,14038200;
> if:1,1,0;
> if:1,1,0;
> if:0,1,-650;
> if:0,0,-650;
> dds:1,28054200;
> dds:0,14038850;
> modulations_list:AM,SAM,DSB,LSB,USB,NFM,FM,DIGL,DIGU,CWL, CWU, CW;
> if_limits:-96000,96000;
> vfo_limits:0,38400000;
> channels_count:2;
> trx_count:2;
> receive_only:false;
> device:SunSDR2PRO;
> protocol:ExpertSDR3,1.8;

EA5SW, 2024-03-13:
I push F12 to TX and change to Thetis to click in MOX and return to RX

EA5SW, 2024-03-13:
In Log4OM the ptt key repeats until you end press and send trx false

> trx:0,false;
> trx:0,false;
> trx:0,false;
< TRX:0,false;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
< TRX:0,true;
> trx:0,true;
> trx:0,true;
> trx:0,true;
< TRX:0,true;

foldynl, 2024-03-13:
Yeah, sorry, you are right. I sent you a half of patch.

Your test clearly shows that Thetis is able to process the TRX command. It means that it is really only needed to remove the TRX command from the initial phase, and it will not affect further processing.

Thank you very much for your patience. I will prepare a patch that will be included in the next release. However, it won't be released for another month unless a critical error arises.

foldynl, 2024-03-13:
The correct patch should be (without testing):

```
diff --git a/ui/MainWindow.cpp b/ui/MainWindow.cpp
index 7baafd4..60934a4 100644
--- a/ui/MainWindow.cpp
+++ b/ui/MainWindow.cpp
@@ -245,7 +245,7 @@ MainWindow::MainWindow(QWidget* parent) :
     /*************/
     /* SHORTCUTs */
     /*************/
-    QShortcut *shortcut = new QShortcut(QKeySequence(Qt::ALT | Qt::Key_Backslash),
+    QShortcut *shortcut = new QShortcut(QKeySequence(Qt::Key_F12),
                                         this,
                                         SLOT(shortcutALTBackslash()),
                                         nullptr, Qt::ApplicationShortcut);
@@ -281,8 +281,8 @@ void MainWindow::keyReleaseEvent(QKeyEvent *event)
 {
     FCT_IDENTIFICATION;
 
-    if ( event->key() == Qt::Key_Backslash
-         && event->modifiers() == Qt::AltModifier
+    if ( event->key() == Qt::Key_F12
+         //&& event->modifiers() == Qt::AltModifier
          && ! event->isAutoRepeat() )
     {
         emit altBackslash(false);
```

EA5SW, 2024-03-13:
> The correct patch should be (without testing):
>
> ```
> diff --git a/ui/MainWindow.cpp b/ui/MainWindow.cpp
> index 7baafd4..60934a4 100644
> --- a/ui/MainWindow.cpp
> +++ b/ui/MainWindow.cpp
> @@ -245,7 +245,7 [...]

Works perfect now.

Many thanks !!!

commit fd9354c1 (2024-03-25): Fixed #327 - TCI: Thetis Connection issue

TRX command disabled during the initialization phase
because if this command is received by Thetis
then Thetis unexpectedly closes the connection.
Expert SDR software does not do this.
closed 2024-03-25 by foldynl