foldynl/QLog #332: Feature: Widget for showing station's profile image from qrz

kyleboyle, 2024-03-17:
It'd be cool to have a widget that could be enabled that would show the entered callsign's QRZ profile image. This is a feature in log4om and aclog and I find generally adds to the enjoyment of contacting new stations. The customizable layout in qlog is ideal for these types of widgets.

[image]

[image]

foldynl, 2024-03-19:
At the moment I see no reason to have this widget. What exactly should it do? Download profile photo from QRZ.com?For this case, there is already the "Web" button, which redirects you via the web browser to the defined URL with the calling profile. Another reason not to implement it is that the given widget will be practically QRZ.com specific widget. 

For now, I leave this issue opened for further discussion, but I think it should not be implemented in the way you suggest.

kyleboyle, 2024-03-19:
Hello sir! I'm not sure I understand your reasoning, as everything informational in the application can be found elsewhere on the web. The point of automatically showing the image as part of the UI is to present it without additional actions required by the operator as they are logging the QSO. As I mentioned in the initial description, I find that showing the profile image increases the enjoyment of logging general dx contacts, and I'm sure many others do as well as it is available in other general logging applications. People usually set it to their a qsl card image or a photo of their station or otherwise. 

> What exactly should it do? Download profile photo from QRZ.com

Yes, the url is in the qrz api response.

Cheers

foldynl, 2024-03-20:
I'm sorry, but I don't see to have a new widget, which will bring nothing more than the display of an image, which can already be displayed today by clicking on the `Web` button. 

I leave this issue opened for possible new ideas about this.

foldynl, 2024-03-21:
QLog should be a community project, let's try a new approach in development - voting. I started a poll for this feature #335

rsaue, 2024-03-28:
As an inspiration, here is how the QRZ.com widget is done in LogHX3.

It is used both as a general search widget, manually entering the callsign, and as a automatic look-up tool for callsigns entered into the logging widget. 

When moving the mouse pointer over the picture, it is shown i real size. 

73 de LB2EG Richard  

[image]

foldynl, 2024-03-28:
Many thanks, Richard. The question is not how to implement it, but whether to implement it. I have some opinion on it, but I don't want it to be rejected when the majority wants to implement it. That's why I created a polling #335 to find out what the community thinks.

foldynl, 2024-04-21:
So the polling was 50:50 for a long time... In the end, however, the result was "it should NOT be implemented"(45:55). But because the new function can be easily turned off, it WILL be implemented.

commit b570409a (2024-04-25): Fixed #332 - Profile Image: Added a new widget to display Callbook profile image
closed 2024-05-03 by foldynl