foldynl/QLog #336: Failing to load grid square for G and EI SOTA summits

sjmelhuish, 2024-03-23:
When I load a SOTAREF into the Details->SOTA field QLog fills in the Gridsquare field. For example, if I enter "F/PO-001" the Gridsquare field is correctly populated with "IN92WS".

However, when I enter e.g. "G/SC-001" the Gridsquare is incorrectly populated with "JJ00AA". This corresponds to lat = lon = 0.0, so is clearly an error. For this example the correct value would be "IO81ED". This applies to all summits which use the British and Irish grids (G, GD, GI, GM, GW and EI).

Why?
I suspect the cause might be `data\Data.cpp`  lines 860-861:

```
        SOTARet.gridref1 = querySOTA.value(6).toDouble();
        SOTARet.gridref2 = querySOTA.value(7).toDouble();

```

(but maybe elsewhere too?)

For most summits in SOTA's summitslist.csv download latitude and longitude are duplicated into gridref1 and gridref2. But for G & EI summits gridref1 contains the alpha-numeric grid reference ("SS 891415" for the example) and gridref2 is blank. Therefore the toDouble() calls will fail. Perhaps also this will cause trouble in the database?

foldynl, 2024-03-24:
Good analysis. Thanks for the info. I looked at it and I don't see why the British and Irish have something different than the rest of the world in Gridref1/2. All we have to do is switch to using Lat/Lon fields. Is it correct?

However, I performed a deeper analysis and found that sometimes the data in Gridref1/2 is more accurate than in Lat/lon. I'm a bit confused. What would you recommend? Is it sufficient to use Lat/Lon information? Or should something more complicated be implemented?

foldynl, 2024-03-24:
It seems that it is sufficient to use lat/lon information. I randomly selected cases of different information (gridref1/2 vs Lat/Lon) and the computed Grid was always the same.

commit ed7cb142 (2024-03-24): Fixed #336 - Failing to load grid square for G and EI SOTA summits
closed 2024-03-25 by foldynl