foldynl/QLog #339: Multiple bandmap window

kaczorks, 2024-03-24:
There is possible to open multiple windows of  bandmap ? I use multiple windows of bandmap for see all band at the same time.

foldynl, 2024-03-25:
it is not implemented and will not be in the near future. Currently, I see following limitations:

1) At the moment I can't imagine how it would be possible to implement it considering the placement in the main window
2) The performance of rendering the bandmap would have to be solved. Keep in mind that DX spot source can be RBN network and there is huge traffic during the contest (around 150k QSOs per hour).

I leave the issue open, because it is a good idea that needs to be more analyzed.

kaczorks, 2024-03-25:
About placement of window i use this in other logger and i put this bandmap windows side by side. I have 5 band antenna and i have open 5 windows of bandmap at the same time on the second screen as external window not in main program window. Many logger can do it like as logger32, log4om.

foldynl, 2025-04-04:
fixed in 0.43