foldynl/QLog #345: Minimal contest support

BG7JAF, 2024-04-07:
I know, I know, QLog won't force on contest. But I think add minimal contest support is necessary. Last month's CQ WPX SSB contest I do log with pen and paper, then I want to add the QSOs into qlog, I found that there is no place to enter contest exchange info, but in the QSO detail dialog there already have a contest tab. If I can enter contest exchange info, I will use qlog to do contest (other contest logger programs are too complex :( ). So Ladislav can you add this contest tab in the main window?

73

foldynl, 2024-04-07:
I often answer this question. At the moment I can answer you as I did on the Groups.io mailing list. However, this does not mean that my decision is final. I tried something new during CQ WPX, but I just test it. Currently nothing specific.
```
Hi Dave,

It's an area I get asked about a lot. I also like to participate in contests, so I would welcome that as well. However, there are several mental barriers preventing its implementation in QLog:

   1) QLog was initially defined as a non-contest log.
   2) While adding a couple of fields for exchange isn't problematic, I fear it won't stop there. We'll likely need to export to several contest formats (EDI, Cabrillo, etc.).
   3) If we support adding fields for Exchange and Export QSO, why not consider adding Voice Macros? I've already received such a request.
   4) If Voice Macros are included, we can't overlook calculating points together with multipliers. Every contest is different therefore it will be needed to create a pseudo-language to define contest rules.
   5)  Exchange can contain more than just a serial number; it also includes a special area, ITU/CQZ zones, etc. We'll need to define Value-Lists for these "special" exchanges and, most importantly, keep them up-to-date.
   6) There are special contests with stages and various restrictions (10-minutes rule etc).

These are just some of the considerations for implementation. If we start with the exchange fields then QLog can become N1MM.

We can open the data discussion, whether I am wrong and if my ideas are too complicated.

```

vazquezrodjoel, 2024-10-14:
I would like also some light contesting option on the log.  Maybe a solution would be to to apply some methods of the N1MM software, meaning the configuration files that applies what fields are needed etc.  I'm not an avid contester so can't talk too much about n1mm or different rules or what not, but if I'm not mistaken n1mm you can create the contest configuration file and then import it to the software.  That way you create the "platform" in a tab next to "My Notes", define the rules for the file, and leave it to the comunity to create a template/configuration file to import for any particular contest.

commit 028a626f (2024-11-24): Fixed #345 - Added minimal contest support
commit 6debd139 (2024-11-24): Fixed #345 - Contest Support: Modified Filters

WSJTX Widget does not show DUPEs - filter is active
DXC Widget does not show DUPEs - filter is active
Alerts Widget shows DUPEs - no filter for dupes
Bandmap Widget shows DUPEs - no filter for dupes
commit ac1121df (2024-11-24): Fixed #345 - Contest Support: Completer for ContestID and added INTL-removal
commit cc187aa4 (2024-11-24): Fixed #345 - Contest Support: QSO filter contains the entire contest ID
closed 2024-11-24 by foldynl