foldynl/QLog #350: Question marks are silently removed from entries

p14e, 2024-04-21:
Hi,

first I must say QLog is an awesome logging program and I like it very much. Keep up the good work!

I'm using QLog 0.34 (commit 5f8bf45bf9) compiled directly from the git-sources.

I just noticed that QLog silently modifies or even completely removes ADIF-tags containing a question mark ("?"). I imported an ADIF-log containing a comment-tag like "foobar ?" and some other field containing only a single "?", then I added some info like gridsquares to several log-entries and exported the full log as .ADI and .ADX. After that I noticed by pure chance that the question-mark disappeared from the comment-tag (it now reads just "foobar ") and the tags with the single question mark were completely gone.

I hope this can be fixed soon.

Thanks & Best Regards,
Patrick

p14e, 2024-05-01:
Hi,

I can confirm commit 9b861669db fixes the issue. I'm looking forward to the next release. :-)

Thanks for fixing it!

Best Regards,
Patrick

foldynl, 2024-05-01:
the next release will hopefully be soon. It will not use commit https://github.com/foldynl/QLog/commit/9b861669dbd451c7e5443dd22fe87a0cdf48d0fe but commit https://github.com/foldynl/QLog/commit/65fc34bbe7a1c8b803ff06683cc4d357d77e57fd . There was an issue under Windows

please be careful when using these devel commits. I don't recommend test them if you don't know what you are doing (more  #361)

commit 9b861669 (2024-04-22): Fixed #350 - Question marks are silently removed from entries

Reword removeAccents to use iconv translit option.
commit db9ad201 (2024-04-22): Fixed #350 - Added libiconv to MacOS configuration
commit 65fc34bb (2024-04-25): Fixed #350 - removeAccents: New algorithm that uses the Perl library Text-Unidecode/iconv approach

The algorithm is also two times faster than the previous one.
This commit also fix an issue:

Fixed #316 - New Translit approach guarantees that Cyrillic characters will be transcribed into ASCII
closed 2024-05-03 by foldynl