foldynl/QLog #353: Yaesu FT-847 hamlib CAT issue

magellan-13016, 2024-04-22:
Hello, 

I have a Yaesu FT-847 and a modern CAT interface supplied by XGGComms and unfortunately I can't use Qlog with my radio because as soon as I click on "Connect Rig", it switches to transmit until I disconnect it.

In fact, the problem is that there are no options in the software to manage how you activate the PTT on the radio. The same options should be added as those in the WSJTX software (VOX, DTR, CAT and RTS).

As far as I'm concerned, the "RTS" option, which is used by my interface, is missing.

I hope that a solution can be found so that I can use this software, which seems to be the best in its category under the Linux environment.

Thank you in advance for taking my problem into consideration.

Best regards, David (F4BPP).

foldynl, 2024-04-22:
temporarily it is possible to use rigctld and set the QLog for using "Hamlib Net RIGCTL" device under Linux. Rigctld has an option that allows you to select the PTT type. Otherwise, if you use Window, you can switch to Omnirig.

magellan-13016, 2024-04-22:
Thank you very much for the answer. I'll give it a try. You said "temporarily"... so does this mean you plan to add the options to the software ?

magellan-13016, 2024-04-22:
Ok, the "Hamlib Net RIGCTL" option is working ! Thanks for the tip! I hadn't thought of it, although I've developed a software to interface the icom PCR-100 with Gpredict using this solution. However, it would be more practical if Qlog could allow you to set the way the PTT is activated via Hamlib.

Anyway, thank you very much! I was looking forward to testing this software!

Best regards, David (F4BPP).

magellan-13016, 2024-04-23:
Until the PTT options are added to the software and in case some people have the same problem, here is the command line to enter in a terminal for the Yaesu FT-847 :

rigctld -m 1001 -r /dev/ttyUSB0 -t 4532 -s 57600 --ptt-type=RTS --set-conf=stop_bits=2,serial_handshake=None

Replace ttyUSB0 by the USB port of your digital interface located in the /dev folder. The baudrate is set to 57600 for my radio so you have to change the value according to your radio. You also have to change the number 1001 which is the value of the radio model.

I hope this will help those who face the same issue.

foldynl, 2024-04-23:
>You said "temporarily"... so does this mean you plan to add the options to the software ?

I don't want to promise anything. Unfortunately, there are many requests, no one is contributing, so it's only up to me and my free time. I cannot estimate whether and when it will be implemented. 

> Replace ttyUSB0 by the USB port of your digital interface located in the /dev folder. 

this is a commonly used approach, but I wouldn't recommend it. It is better to use a device from `/dev/serial/by-id/`, so these links do not change over time. ttyUSB links may vary.

magellan-13016, 2024-04-23:
Thanks for your reply. To be honest, I looked in your profile for a link to make a donation and couldn't find one. I realise that I'm asking you to do some extra work, and as a programmer, I think it's normal that all work should be rewarded. So give me a Paypal link and I'll make you a donation straight away.

As far as ports are concerned, I switched to persistent names a long time ago. So each device is identified and with rules in udev, I give it a unique name. In my case it's ttyUSB.DIGIMODE and no matter what order I plug in my USB devices, ttyUSB.DIGIMODE always points to my interface. 

Thanks anyway for the tip about "/dev/serial/by-id/" which I didn't know.

magellan-13016, 2024-04-24:
Thanks for your reply. I don't know what you're referring to with "PR" but although I don't have priority over other more important issues, I'd still like to make a donation to the developers. Just give me a paypal link and I'll take care of the rest.

foldynl, 2024-04-25:
PR = Pull Request - sending QLog changes via Github, in the other word, you will contribute by changing the code so that QLog can improve faster

>> Just give me a paypal link and I'll take care of the rest.

I have no link for it. I am non-profit project. I come from the old school, where the emphasis was on the pleasure of enhancing software. Money wasn't a driving force in speeding up development in this scenario.

magellan-13016, 2024-04-25:
Don't worry, my aim was not to boost your productivity with money, just to give you a helping hand as a thank you for the work you've already done. As far as I'm concerned, I'm just like you. On my site (www.f4bpp.com) I'll soon be putting several pieces of software I've coded online free of charge, including one that lets you use an Icom PCR-100 receiver with Gpredict and QSSTV to receive SSTV images transmitted by the ISS while correcting for the Doppler effect. The same applies to the FT-847, along with an application for Android smartphones... free of charge.

Translated with DeepL.com (free version)

foldynl, 2024-05-27:
@magellan-13016 , please, are you able to compile the commit above whether it is working for you? I don't want to release it without being able to test it

magellan-13016, 2024-05-27:
I don't know how to compile anything except some simple Python programs. I propose you to compile the software and give me the deb package so I can test it and come back to you for the feedback.

foldynl, 2024-05-28:
Do you already have QLog installed? What distro are you using?

magellan-13016, 2024-05-28:
Yes I have Qlog version 0.35.2 and I use Kubuntu 22.04 based on Ubuntu.

foldynl, 2024-05-29:
Perfect. I will perform some internal tests and then I will only provide you a binary with instructions. I'm also using 22.04, so we have the same versions of binaries/libraries and it is not necessary to prepared a special package.

foldynl, 2024-05-30:
Please, follow [link](https://drive.google.com/file/d/1e-T4_hJH6UxaKxjvsli92X9eFAwNS6nf/view?usp=sharing) to download an experimental binary of QLog that support PTT Type change.

Sequence to test
1) Download the binary
2) Add execute bit for the binary - `chmod u+x qlog`
3) Run the binary with a special parameter - `./qlog -n test`
4) The special parameter ensures that a new QLog profile is created. In other words, your current QLog database will not be affected by the test. Your data will remain unchanged. The 'n' parameter will create a new instance of QLog, with no profiles and no QSOs, but your original database with your QSOs will remain intact.
5) Define a new station profile as usually 
6) Define a rig profile
7) Test your rig connection

please be careful when running testing QLog. There must be the "n" switch. Without it, your current QLog database would be upgraded and you wouldn't be able to run your current QLog.

magellan-13016, 2024-06-02:
Many thanks for your work Ladislav ! 
When I run Qlog with "./qlog -n test" I get this warning : 
[WARNING ] [0x7f9c711da640]  [             ] => "rig_settings_get_path: path=/home/master/.config/hamlib_settings" [void Rig::__openRig():../QLog/rig/Rig.cpp:329] 
And my problem is still here : When I connect the rig, it switches to PTT until I disable it.
Here is my setup :
[image]

Once again many thanks for all your efforts !

foldynl, 2024-06-02:
It is strange.Please could you send me your WSJTX setting which is working for you?

magellan-13016, 2024-06-02:
[image]

magellan-13016, 2024-06-02:
[image]

foldynl, 2024-06-02:
Unfortunately, without your help it will not be possible to finalize it. I have no idea where the error could be and I have no way to test it.

Please, could you  experiment with `rigctl` (it is a hamlib's tool), which controls the rig. Based on your setup from WSJTX, please try to run `rigctl -vvvv -m 1001 -r /dev/ttyUSB.DIGIMODE -s 57600 -p /dev/ttyUSB.DIGIMODE -P RTS`

what is the effect Is it the same behavior as in QLog? If so, can you experiment with rigctl to be able to control your rig? Then send me the resulting command back. 

Maybe, could you test also this one:
`rigctl -vvvv -m 1001 -r /dev/ttyUSB.DIGIMODE -s 57600 -p /dev/ttyUSB.DIGIMODE -P RTS --set-conf=ptt_share=1`

does it work or not ?

commit 5d732969 (2024-08-23): Fixed #353 - Added RTS/DTR PTT Type support
closed 2024-08-29 by foldynl