foldynl/QLog #354: No sorting options and distance not calculated in logbook

magellan-13016, 2024-04-23:
Hello,

First of all, I'd like to point out that this logbook is the best I've found on Linux. Now that I've adopted it, I'd like to take this opportunity to mention two shortcomings: 

1. The first is that there is no sorting option. I think it's essential that you can click on a column header to have the table sorted according to the criterion of your choice (date, callsign, mode or distance).
2. The second is that if you decide to display the distance in the log, it's not calculated and it's up to the user to enter the distance themselves, which other software such as WSJTX or Gridtracker are perfectly capable of calculating.

With the addition of the PTT mode for Hamlib users that I mentioned in a previous post, this software would be perfect for me.

Thank you for considering my request and many thanks again for this fabulous software.

Best regards, David (F4BPP).

foldynl, 2024-04-23:
please don't combine multiple things into one Issue report. This is not a mailing list, but an Issue Tracking  system. Each issue would have its records here.

> The first is that there is no sorting option. I think it's essential that you can click on a column header to have the table sorted according to the criterion of your choice (date, callsign, mode or distance).

it's a nice-to-have file. I have to wonder if this is possible in terms of performance. This is not just about the implementing itself, but about the impact on performance during various operations.

> The second is that if you decide to display the distance in the log, it's not calculated and it's up to the user to enter the distance themselves, which other software such as WSJTX or Gridtracker are perfectly capable of calculating.

QLog calculates the distance if it knows both locators, at least in the form of XXYY. You didn't write an exact description of the situation under which and what happens. That's why I can't give you an exact answer here.
However, it definitely does not let QLog fill in the user's distance.

magellan-13016, 2024-04-23:
Sorry Ladislav, I've combined the two problems into one post as they both concern the Logbook table.I'll be more rigorous in the future.

About the sort options, I don't think it would impact the performance because Qlog is very well programmed ;-)

About the distance, after some investigations, I just discovered that when a contact is added to the log, the gridsquare is not filled so the software can't calculate it. In fact, this data is not captured by Qlog on qrz.com but it should like it is the case for the name, the country, etc.

magellan-13016, 2024-04-23:
After some new investigations, there is something I don't understand. I just made a contact with a Germany based station and Qlog was able to show its location on the "Online Map" with the grid overlay. So, why the gridsquare of the station is grabbed but not filled in the contact field ? I just copy the gridsquare of the staion in the map manually and I'm wondering why Qlog is not able to do the same thing itself.

foldynl, 2024-04-24:
QLog calculates a distance only if grid is specified. If there is no grid, no calculation can be performed. Grid can be obtained from the operator or from the Callbook. 

Displaying the route on the "Map" is another task. The map shows the route even if no grid is specified. If the grid is not specified, then the Map shows only the "approximate" position obtained from the DXCC information. This position is very approximate and cannot be used to calculate the distance, because it is the position of the (center of) Country. This feature was added to make it possible to see where the country is placed without knowledge of the grid - useful for small countries or Islands.

magellan-13016, 2024-04-24:
I recently got in touch with AO75MA and found his grid on qrz.com : IN80dj. So I'm wondering why Qlog can't retrieve it from qrz.com like the other data. I think that if WSJTX can retrieve this grid and calculate the distance, Qlog should be able to do the same. But maybe I'm wrong. I haven't put in the qrz.com API key because I don't have one, so I'm wondering if that's why Qlog can't retrieve it and what are the advantages of this API key?

Translated with DeepL.com (free version)

magellan-13016, 2024-04-24:
Anyway I really miss the possibility to sort the Cluster chart by frequency and mode. Same thing for the Logbook that I often want to sort by call, country, date or mode that is not missing in my old software, jLog.

foldynl, 2024-04-25:
>> I haven't put in the qrz.com API key because I don't have one, so I'm wondering if that's why Qlog can't retrieve it and what are the advantages of this API key?

From [QRZ Spec](https://www.qrz.com/docs/xml/current_spec.html): "Access to this service requires user authentication through the use of a valid QRZ.COM username and password. While any QRZ user may login to the service, an active QRZ Logbook Data subscription is required to access most of its features. Non-subscriber access limits the data fields that are returned and is primarily intended for testing and troubleshooting purposes only."

You need XML-subscription (or higher) to get all fields from QRZ. It is not a QLog policy, it is QRZ's policy.

>> Anyway I really miss the possibility to sort the Cluster chart by frequency and mode. Same thing for the Logbook that I often want to sort by call, country, date or mode that is not missing in my old software, jLog.

Sorting is not a user-interesting function in Clutter, because it can confuse users - QSOs change differently there, because their order changes dynamically as QSOs arrive - tested many time, especially in WSJTX Window. Another thing is that I don't see the benefit of having QSOs sorted by Callsign in the DX Cluster window. If you need to filter, that's what QLog's QSO filtering provides. Keep in mind that the source of the Spots can also be RBN, which at the time of contest generates 25 and more QSOs per second. In this case, it would be irresponsible to give the operator the possibility to somehow sort QSO in the DXC window.

magellan-13016, 2024-04-25:
OK. Thanks for the explanations but I still don't understand why, despite I don't have an API Key, Qlog is able to display the station on the online map with a line between the two stations but can't calculate the distance. Sorry to bother you with that. I'm sure that if it does not make it, it's because it can't but I still don't understand why.

aa5sh, 2024-07-31:
I know this is an old topic but it may be the map is drawn just to the center of the DXCC entity if the grid is unknown.  The distance for the contact is only calculated if the gridsquare is known otherwise it would not be nearly accurate.

foldynl, 2025-04-05:
the changes have already been implemented. I'm closing this issue. If necessary, please open with a specific request

closed 2025-04-05 by foldynl