foldynl/QLog #355: Bandmap Enhancement

magellan-13016, 2024-04-23:
It would be nice if, when you hover over a station callsign with the mouse in the Bandmap, the frequency was also indicated in the label displayed just above it. Actually we need to click on the station to display its frequency but in the same time, this frequency is set to the transceiver.

David (F4BPP).

foldynl, 2024-04-24:
What is the use-case for this? In what case do you need this information? You know the approximate frequency, you need the exact if you want to tune it, and that's as easy to click on the callsign.

magellan-13016, 2024-04-24:
Because I faced a problem when a station made a QSY. I switched to the new frequency but losses its callsign and if the frequency was displayed on the label, it would have been easier for me to retrieve it because there were several stations for the same country close to each other on the QRG. But I admit that this development is not a priority. It's jus an improvement suggestion for later.

commit 65f3e8bb (2024-04-30): Fixed #355 - Bandmap: Added Callsign/Freq/Mode to Tooltip
closed 2024-05-03 by foldynl