foldynl/QLog #360: Easier band/freq insertion eithout CAT

mcogoni, 2024-04-26:
Hi,
I didn't find any way to easily change the band/frequency with the keyboard when I use a radio without CAT.
Maybe it's something that could easily be added...

marco

foldynl, 2024-04-26:
I don't understand exactly. You are using a rig without CAT. So how do you have it connected? What is your RIG Setting? Best to send a screenshot.

Otherwise, a shortcut for switching the band. Hmmm, QLog should be the logging software. You should use specialized software to control the Rig. But it's not a bad idea.

mcogoni, 2024-04-26:
Sorry, I did not explain the situation well.
What I mean is that sometimes I use two radios almost at the same time, one is connected to CAT, but the other one is an old radio and it would be nice to manually switch band in QLog. I tried using PGUP-DOWN and it increases/dec the freq. as does CTRL UP DOWN, maybe the PG UP DOWN, while within the freq field, could simply change band within the ones allowed by the user... What I mean is something like 14.200 -> 18.100 -> 21.200 etc.
Does it make sense to you?
Of course when the CAT is active the behavior must remain as it is.

foldynl, 2024-04-28:
Oh so. I see now.

foldynl, 2024-06-08:
Implemented in v.0.36.

closed 2024-06-08 by foldynl

mcogoni, 2024-06-08:
Thanks! 73