foldynl/QLog #361: Global profile lost upon addition of ITU and CQ region in 0.35 dev branch

rsaue, 2024-04-28:
Hi,
I have been following the testing_0.35 branch lately. All good until tonight, with the ITU/CQ zone changes. When starting Qlog, the Settings - Station tab was empty. All other settings; equipment, sync, etc. was still there. Added the station profile data, and saved, the station data is lost/not shown on restart... 

73 de LB2EG Richard

rsaue, 2024-04-29:
Hi,
My only intention was to report a possible config fault before the general relase of version 0.35…  I am very happy that you keep the development branch open. The the few among us who compiles from it, knows there can be some hazards involved  :) 

Keep up the good work!

73 de LB2EG Richard

foldynl, 2024-04-29:
good to know that someone is checking me ;-) I'm a little nervous about the recent changes, so testing phase must be longer than usual

closed 2024-05-09 by foldynl