foldynl/QLog #364: wca world castel award

ik3ocd, 2024-05-05:
good afternoon Ladislav, first of all thanks for the good news on qlog, little by little it becomes nice and friendly and usual, some considerations on the wwff , pota, etc.. during my activity as a researcher on portable stations I see the lack of management as indicated above on wca world castel award activities, I hope you will consider this insertion on qlog in the future. Kind regards Flavio ik3ocd

foldynl, 2024-05-05:
Flavio, there are a large number of these special activities and it is not possible to integrate all these special activities together with their rules. Even the ADIF format does not have defined fields for all these activities. Currently, QLog only contains widely used activities.

For all other special activities, the fields SIG and SIG Info are usually used, where SIG holds the type of activity (WCA etc.), and SIG Info holds the specific ID of the activity. It is necessary to familiarize yourself with the rules of your activity and how QSOs should be recorded and reported.

ik3ocd, 2024-05-05:
Thank you for your consideration regarding this and the function of the SIG and SIG Info fields. thank you, see you soon .

closed 2024-05-06 by foldynl