foldynl/QLog #367: Can't connect to k3ng keyer

BG7JAF, 2024-05-07:
k3ng keyer can simulate as winkey v2, so I tried to connect QLog to a k3ng keyer using 1200 baudrate, but the connection failed. I have also try n1mm+ and fldigi and both them can work with that k3ng keyer, so I think maybe it's QLog's problem?

73

foldynl, 2024-05-07:
I am also using the K3NG Keyer as WinKey v2 Keyer without any issue. It is one of my test cases to use K3NG as Winkey. Your setting seems to be correct. It is important to have setting described here:

foldynl, 2024-05-07:
Please send me your K3NG Keyer version?

BG7JAF, 2024-05-08:
Hi foldynl,

My k3ng keyer's version is  "2024.03.20.2239"
After disable automatic software reset(I just add a big capacitor between RESET and GND), I can connect k3ng keyer with QLog now.  Thank you very much!

73

closed 2024-05-08 by BG7JAF