foldynl/QLog #370: cannot get Rig's network configuration to stay saved

n1kdo, 2024-05-15:
Hello.

New build of QLog, checked out today from commit ee5d16d.  

Built today on Fedora 40.  QLog 0.35.1, Qt 5.15.13, HamLib 4.5.5.

When I set up my rig (a K4) in the File->Settings->Equipment->Rigs panel, I set the rig up for

Port Type: Network
Host Name: (redacted)
Port: 9200

Click Modify, the profile name disappears, the hostname disappears, the port reverts to 4532.

Double-click on "k4" in the profiles box (left column) -- the field values re-appear, and are proper.

Click Modify, they disappear again, click Save, the panel closes.  

If I bring up the panel again (File->Settings->Equipment->Rigs) and double-click on the rig profile name "K4" in left column, the "Port Type" is set to serial. 

As far as I can tell, I cannot save the rig configuration to use network. QLog shows the rig name in red, it does not connect.

Thanks, 73,

Jeff n1kdo

n1kdo, 2024-05-15:
I don't know if this will be helpful:

```
sqlite> .open qlog.db
sqlite> .headers on
sqlite> select * from rig_profiles;
profile_name|model|port_pathname|hostname|netport|baudrate|databits|stopbits|flowcontrol|parity|pollinterval|txfreq_start|txfreq_end|get_freq|get_mode|get_vfo|get_pwr|rit_offset|xit_offset|get_rit|get_xit|default_pwr|get_ptt|qsy_wiping|get_key_speed|assigned_cw_key|key_speed_sync|driver|dxspot2rig
K4|2047||(redacted)|9200|0|0|0.0|||500|1.79999995231628|54.0|1|1|1|1|0.0|0.0|0|0|100.0|0|1|1| |0|1|0
sqlite> 

```

I did replace the host name with (redacted) -- nobody needs to know.  It was the correct hostname.

foldynl, 2024-05-16:
Many thanks for detailed description. It is replicable on the other platforms.

n1kdo, 2024-05-18:
Hello,

I can confirm the fix in the 'bugfix-370' branch works for me.  Thank you!

73,  Jeff n1kdo

foldynl, 2024-05-19:
Many thanks for your confirmation.

commit 3a05273c (2024-05-21): Fixed #370 - Fixed Network Rig configuration is not saved