foldynl/QLog #375: Spot Last QSO Feature

Taffman, 2024-05-23:
'Spot Last QSO' doesn't appear to be working in Windows version 0.35.2.

foldynl, 2024-05-23:
Linux is working....I have to test it in Windows.

foldynl, 2024-05-23:
just adding what I sent via Group.io:

>> I just test it and everything works as expected. Apologies to everyone for the self-spotting on DXC.
>>the idea behind the Spot Last QSO button is that:
>>1) you enter a new QSO
>>2) press the button
>>3) it fills the "Command" for the DX Cluster according to the last QSO
>>4) here you can add a remark or edit the command
>>5) press enter.
>>That is all. The spot will be sent.

foldynl, 2024-05-23:
Tested many times in Windows version 0.35.2 and it works as expected. please, could you described your issue in more details. Screenshots would welcome.

foldynl, 2024-05-30:
discussed via email. There was a misunderstanding. Last QSO is not supported for imported QSO via ADIF import.

closed 2024-05-30 by foldynl