foldynl/QLog #376: Migration from 0.26 -> 0.35.2 database corruption ?

CutePotatoDev, 2024-05-28:
I upgraded from 0.26 to 0.35.2. After I started new version my logbook is empty. But user profile looks like loaded.

Noticed that the country filter dropdown doesn't contain any country. Dropdown is empty.

Also when uploading adx file with backup qso duplicates is detected. Then i guess my qso's still in database.

foldynl, 2024-05-29:
what OS are you using? In case of Linux, do you use installation package or compile it from source code? 

I tried your configuration on Linux and I didn't notice any issue.

CutePotatoDev, 2024-06-03:
MacOS 10.13.6, compiling from source.
After some check problem appears between versions v0.33.0 -> v0.33.1.
How i can note, only significant change between this versions is qt upgrade.

Because of my slightly outdated system and self build software i don't see any necessity for further investigation.
I try to solve this somehow myself. Upgrade my qt and check or it helps. Stick to last working QLog version. Or just it's already time to get newer computer.

Thank you for your time.

closed 2024-06-03 by CutePotatoDev