foldynl/QLog #377: Callbook search doesn't work - 0.35.2

dimarmnes, 2024-05-29:
Version: Qlog 0.35.2

Os: Windows 10 and Linux Manjaro (Flathub version)

Issue: After entering the callsign in new QSO and pressing enter, nothing happens.
Tested with QRZ and HamQTH, the callbook search doesn´t work

foldynl, 2024-05-29:
what color is the icon next to Callsign?
[image]

dimarmnes, 2024-05-29:
it's green and say "Callbook search is active".
The strange thing is that it happened on both Linux (main) and Windows that I installed to test. The only thing in common is that in Windows I imported the .adx file generated in the Linux version.
On my main PC (linux), I downgrade to the previous version and it worked again.

TecnoTalarn, 2024-05-31:
In my case, it work, but very slow. Also impact to change frequency when run via bandmap.
Normally a search or change  freq run in millisecons, but now i see 2 or 3 seconds to change.

commit d7bc8214 (2024-06-04): Fixed #377 - Callbook query does not work
closed 2024-06-07 by foldynl