foldynl/QLog #379: FT8 Contacts Not Synced to QRZ

jmcblane, 2024-05-31:
When FT8 Contacts are imported, either by QLog automatically or manually, they are not synced to QRZ regardless of the QRZ syncing status. Attempting to upload these contacts to QRZ just results in being told there are no new contacts to upload to QRZ.

foldynl, 2024-05-31:
what is the QRZ Upload Status value for the impacted QSO(s)? Can you export one of these QSOs and send me an ADIF record?

jmcblane, 2024-06-01:
I tried changing the upload status to each option and uploading to QRZ
with the same result. It is like it doesn't recognize that the QSO
exists in the log at all or something. Below is the ADIF record of one
such QSO. Mind you, I manually changed it to Y for QRZ upload status
after manually adding the contact to QRZ from the QRZ web page.

<call:6>TI5VWR
<rst_sent:3>-06
<rst_rcvd:3>-11
<freq:6:N>14.074
<band:3>20m
<mode:3>FT8
<name:22>Luis Fernando Montalto
<qth:17>Altamira, Guacima
<gridsquare:6>EJ79UX
<dxcc:3>308
<country:10>Costa Rica
<cont:2>NA
<cqz:1>7
<ituz:2>11
<pfx:3>TI5
<qsl_rcvd:1>N
<qsl_sent:1>N
<lotw_qsl_rcvd:1>N
<lotw_qsl_sent:1>N
<tx_pwr:2>45
<band_rx:3>20m
<distance:4>4864
***@***.***
<eqsl_qsl_rcvd:1>N
<eqsl_qsl_sent:1>N
<freq_rx:6>14.074
<lat:11>N009 58.752
<lon:11>W084 17.520
<my_city:6>AUBURN
<my_cnty:9>CA,Placer
<my_country:13>United States
<my_cq_zone:1>3
<my_dxcc:3>291
<my_gridsquare:6>CM98KW
<my_itu_zone:1>6
<my_lat:11>N038 55.309
<my_lon:11>W121 05.378
<my_name:13>JACOB MCBLANE
<my_state:2>CA
<qrzcom_qso_upload_date:8>20240531
<qrzcom_qso_upload_status:1>Y
<station_callsign:5>K6AVB
<qso_date:8:D>20240531
<time_on:6:T>055100
<qso_date_off:8:D>20240531
<time_off:6:T>055100
<app_qrzlog_logid:10>1104824977
<app_qrzlog_status:1>N
<eor>

jmcblane, 2024-06-01:
Sorry for the double response, but I made a quick FT8 contact which
QLog pulled in automatically, and here is the result:

<call:6>KF4DGS
<rst_sent:3>-06
<rst_rcvd:3>+00
<freq:7:N>21.0743
<band:3>15m
<mode:3>FT8
<name:13>Standon D Lee
<gridsquare:4>EM63
<dxcc:3>291
<country:13>United States
<cont:2>NA
<cqz:1>5
<ituz:1>8
<pfx:3>KF4
<qsl_rcvd:1>N
<qsl_sent:1>Q
<lotw_qsl_rcvd:1>N
<lotw_qsl_sent:1>Q
<tx_pwr:3>100
<a_index:2>12
<distance:17>3090.636389051466
<eqsl_qsl_rcvd:1>N
<eqsl_qsl_sent:1>Q
<k_index:1>1
<my_antenna:8>40m EFHW
<my_city:14>Auburn, CA USA
<my_country:13>United States
<my_cq_zone:1>3
<my_dxcc:3>291
<my_gridsquare:4>CM98
<my_itu_zone:1>6
<my_name:13>Jacob McBlane
<my_rig:7>IC-7300
<sfi:3>179
<station_callsign:5>K6AVB
<qso_date:8:D>20240601
<time_on:6:T>025945
<qso_date_off:8:D>20240601
<time_off:6:T>030045
<eor>

foldynl, 2024-06-01:
`qrzcom_qso_upload = Y` means "the QSO has been uploaded to, and accepted by, the online service" ( more details in [ADIF](https://adif.org/314/ADIF_314.htm#QSOUploadStatus_Enumeration) specification)

`qrzcom_qso_upload = N` means "do not upload the QSO to the online service" ( the same specification).

If you want to send QSO where `qrzcom_qso_upload = N` then use QRZ Upload dialog and enable `Include QSO ....`

[image]

If `qrzcom_qso_upload` is missing in ADIF or the value is empty then the QSO will be sent to QRZ. 

Did you tried to upload your QSO with KF4DGS?

jmcblane, 2024-06-01:
Thanks for the crash course. This confirms what I thought was going on. However; nothing I do, including the suggestions you have given, will get the software to recognize the QSO and send it to QRZ. Marking it as N and checking the box to include those doesn't work. Blank doesn't work. And it's only with QSOs imported from WSJTX, whether the program imports them automatically or if I import the ADIF manually, same problem.The QSO with KF4DGS will not sync to QRZ through QLog.JACOB MCBLANE

image.png (view on web)

foldynl, 2024-06-02:
Jacob, I'm sorry you're having these problems, but I can't replicate your problem. I imported both QSOs into my QLog. And both QSOs can be selected for Upload to QRZ. Of course, after changing Y->N. The only problem I see is that you log one QSO with a short locator, and then the second QSO with a long locator. But it doesn't affect the choice. You just have to send the QSOs with two selection of the grid. see short video below.

[image]

jmcblane, 2024-06-02:
I found the issue. I was not selecting the other shorter prefix before asking it to upload. That was my problem.I sincerely appreciate your time and effort, and maybe this will help someone else who runs into the same issue. I expected it was searching all the QSOs, and not only checking those of the grid square selected on the upload box.JACOB MCBLANE

jacob.gif (view on web)

closed 2024-06-03 by foldynl