foldynl/QLog #387: export brand qso y or n

ik3ocd, 2024-06-08:
good morning ladislav, a question about exporting qso for printing labels, is it possible to use the function during export to mark the chosen qso as qsl sent in this case as Y sent or leave as N sent? it would be a nice automation  🤔 in the case of several qso during export. thanks for your consideration. 73'

foldynl, 2024-06-08:
QLog already contains the function. If you export the QSO and the Export type is QSLs then you have an option to mark the QSOs as "Sent" or leave them as they are. 

[image]

Keep in mind two things:

1) QSLs Export - QSOs are exported that have set conditions as specified in the filters, and at the same time, the QSL Sent status is Requested (`R`) or Queued (`Q`).
2) ADIF specification defines QSL Sent with the [enumeration](https://www.adif.org/314/ADIF_314.htm#QSLSent_Enumeration) where `N` means "do not send". I am not sure if you want to mark the QSOs as QSL Sent = `N` for later processing.

ik3ocd, 2024-06-08:
Thanks again for your comprehensive answer and help Ladislav. 73s of IK3OCD Flavio

foldynl, 2024-06-08:
Can I close this issue?

foldynl, 2025-04-05:
Closing the issue

closed 2025-04-05 by foldynl