foldynl/QLog #388: QRZ Lookup Not Including Full Name in 0.36.0

clayton-workman, 2024-06-09:
This appears to have started happening in 0.36.0.

QRZ lookup isn't including the full name.

Example 1:
[image]
[image]

Example 2:
[image]
[image]

Example 3:
[image]
[image]

I'm using 0.36.0 on Fedora 40.

Thanks,
Clayton, KV4CW

dimarmnes, 2024-06-09:
It's working to me.

Qlog 0.36.0

clayton-workman, 2024-06-09:
Thanks. I'll try to the flatpak and a fresh compilation to see if I see different behavior.

Clayton

foldynl, 2024-06-09:
It also works for me

[image]

clayton-workman, 2024-06-09:
I think I see is going on. 

Everything works ok, if....
1) I use the flatpak version (which has it's own config and log file location, so it starts with fresh files.)
2) I use the 0.36 rpm, but start with a new log.

I now see why using a new log is important. It is because QLog will remember any changes you make to the name in the log when doing a lookup. Instead of inserting the lookup value it will keep the previously used value in the log. I apparently have an issue with older log file entries that were imported into qlog and the complete names were not imported. When I try to do a lookup on the callsign after import, it will use those incomplete names in the log instead of what is in QRZ. I'm not sure if this is intentional? It would be nice to be able to have qlog do a lookup and correct these entries instead accepting the previously logged value no matter what.

Examples (all version 0.36.0):
Flatpak version with fresh log file works OK:
[image]
RPM version using my old log file:
[image]
RPM version with fresh log file:
[image]

Ok... now with the fresh log file, I will modify the name by deleting everything after 'WILLIAM C "crai' before saving to the log.
[image]
[image]
Now future lookups will just include the name as saved in the log instead of the name on QRZ:
[image]

So, I'm not sure if this is intended. If so, I will close this issue. I would like the ability to update entries with lookups. (However, I do appreciate that editing a record and doing a lookup does not over-write the grid square in case the QSO was actually at a different location than the grid square published for the person's QTH on QRZ.)

This may mean I need to look at the ADIF import and see if there was an issue importing names as well, but I'll post that as a separate issue if I find anything there.

clayton-workman, 2024-06-09:
On second thought (regarding the import).... these are log entries from gridtracker communicating with qlog (not adif import). Apparently, the full name isn't getting interpreted correctly. Again, I'll look further and log that as a separate issue if I find that it is an issue with qlog.

clayton-workman, 2024-06-09:
Here are some more details of how I am getting the lookup to not work correctly....

When a contact is being logged from WSJTX, the name is not stored - it is blank.
For example:
[image]

I have grid tracker setup to forward WSJTX messages on 2238 and QLog is connected to that port. When I initiate contact, the details show up in window before being saved in the log. The name is queried correctly. It's just that when the contact is saved, that the name is missing.

In order to fill in the name details, I have been opening the log entry, editing it, and then doing a lookup:
[image]

It is this lookup that is not pulling in the complete name. When I save the contact, any future lookups of that callsign will used the stored incomplete name instead of the looked up name is detailed in the previous post. 

I hope this adds more clarity to the issue.

73
-Clayton

foldynl, 2024-06-10:
thanks for the very detailed description of the situation. Great job. This is exactly what I always need as an input - example, cases, situation before and after.

Yes, now I understand what the problem is. The first case is not so serious, because it is a "wanted" functionality to take information from previous connections, but your second case is more serious and I need to go back to the beginning and try to rework it.

dimarmnes, 2024-06-10:
I have the same setup as Clayton, only instead of WSJTX, I use JTDX. Qlog also listens to port 2238 with the information provided by Gridtracker. However, I just tested for the same error and found no problems. Qlog performs the search correctly and saves the contact's full name.

73, LU1IDC, Diego

clayton-workman, 2024-06-10:
Hi Diego, make and save a dummy contact with a valid callsign but remove the name from the name field. (Save it with an empty name.) Now double-click to open the QSO record, press Edit and then lookup. The name will fill in but will be partial. This is the primary issue.

Clayton

dimarmnes, 2024-06-11:
Now I understand what the error you are referring to is. I tried what you told me and the error also occurs for me.
What I found strange is that Qlog did not save the contact name sent by WSJTX/Gridtracker.

Diego

closed 2024-06-20 by foldynl