foldynl/QLog #389: Authentication to QRZ.com should be done over POST with form data.

ericmoritz, 2024-06-09:
https://github.com/foldynl/QLog/blob/9dc6db6b69fa79e6fc3ebd8ada60843b5c83829e/core/QRZ.cpp#L275

Hi, I was writing an application to query callsigns using `xmldata.qrz.com`. While reading the docs I saw that they suggest using a query string for the username and password. This is considered a bad practice because it potentially leaks the account's password in logs and other intermediaries likes caches and proxies.

https://owasp.org/www-community/vulnerabilities/Information_exposure_through_query_strings_in_url

The good news is that they support making requests over POST using `application/x-www-form-urlencoded`. I would recommend making requests using POST to remove the potential that a person's application password might leak between qlog and `xmldata.qrz.com`.

Sample request:

```
POST /xml/current/
Host: xmldata.qrz.com
Content-Type: application/x-www-form-urlencoded

username=xx1xxx;password=abcdef;agent=q5.0
```

The risk is probably low because they're using HTTPS and query strings are not visible to intermediaries as long as TLS is active. Though any intermediary proxy between TLS termination and the API service will see the password in the query string. Not knowing how the URL might be logged makes me nervous, so I'd recommend switching the POST whenever possible to mitigate this risk.

foldynl, 2024-06-09:
Thanks for the warning. I'm glad that someone audits the code from the point of view of security as well.

foldynl, 2024-06-09:
Issue also occurs in case of eQSL, HamQTH and LoTW. Unfortunately, eQSL and HamQTH seem to require GET. LoTW is down so I can't download the API spec to confirm whether POST is supported.

commit ec4c0384 (2024-07-01): Fixed #389 - QRZ auth over POST with form data
closed 2024-06-20 by foldynl