foldynl/QLog #393: QLog Snap packaging

AsciiWolf, 2024-06-18:
QLog already has a Flatpak, however Flatpak is unsupported on Ubuntu. It would be great to provide a [Snap](https://snapcraft.io) of QLog that would be available for easy installation using the "App Center" GUI on Ubuntu (unlike the PPA package that is hard to install by regular users).

foldynl, 2024-06-18:
>>> however Flatpak is unsupported on Ubuntu
flatpak doesn't have GUI integration, but it is supported. I have a virtual machine with Ubuntu for my QLog flatpak tests

Otherwise I agree with you.

AsciiWolf, 2024-06-18:
It sadly has way more issues than just missing GUI (App Center) integration. The flatpak package itself is often very outdated (even contain unfixed CVEs), there are some weird bugs and other problems with system integration etc.

AsciiWolf, 2024-06-18:
By the way, CQRLOG is broken (and KLog is not very good) on Ubuntu since 20.04, so QLog Snap would probably be highly appreciated by many Ubuntu using hams. :)

foldynl, 2024-06-18:
>>> By the way, CQRLOG is broken (and KLog is not very good) on Ubuntu since 20.04

Therefore, I started to develop QLog ;-) I am a former CQRLog user.

I am snap newbie. Currently, I am not able to say what does it means to prepare SNAP. But maybe it's time to think about it. I'm finishing the next release and I'll try to focus on that in the next one. Or do you have any experience?

AsciiWolf, 2024-06-18:
I have some very basic experience (have made Snap for one game). But I will try to provide help. :-)

AsciiWolf, 2024-06-20:
As an example, you can take a look at the only two Snaps that I made:

https://bazaar.launchpad.net/~asciiwolf/autoremover/trunk/files
https://git.launchpad.net/tremulous-snap/tree/

The first one is very old, but is pretty simple and uses a Makefile. The second one is just using a pre-compiled binary archive.

[Here](https://snapcraft.io/docs/c-c-applications) is also a nice basic tutorial.

I am not sure how are AppStream metainfo files handled by Snaps. I think that they are not compulsory and may also be (partially) overwritten by the metadata specified directly in the snapcraft.yml file + at the admin section of the snapcraft.io website.

AsciiWolf, 2024-06-20:
Here are more community maintained Snaps that you can take a look at:

https://github.com/orgs/snapcrafters/repositories?type=all

AsciiWolf, 2024-06-21:
@foldynl So, I did take a look at it and have made this:

[qlog-snap.zip](https://github.com/user-attachments/files/15933527/qlog-snap.zip)

It is still very incomplete and won't even compile, but I think that it is a good start and we can use it as a base for the Snap.

I am not sure about the automatic updates of that Snap though. They are somehow supported by the Snap build pipeline as far as I know, but this probably applies only to the main app, not its dependencies.

AsciiWolf, 2024-06-27:
By the way, I am not sure how Snap handles multiple desktop files. As an alternative, we could add something like this to the main qlog.desktop file:

```
[Desktop Action TQSL]
Name=TQSL
Exec=tqsl
```

This should add "TQSL" as a right click option to the "QLog" desktop entry. It should work fine in GNOME and probably also in KDE, but I am not sure about the other desktop environments.

foldynl, 2024-06-27:
To be honest, I managed to create a snap with QLog, but there are many issues than just TQSL. Task postponed.

AsciiWolf, 2024-06-27:
Ok, fair points. Could you at least share your Snap manifest so I could take a look? Thanks!

foldynl, 2024-06-27:
I don't know if there is anything to share. It's a merge of several examples and my experiments. It is possible to run QLog but snap is not optimized, the password manager doesn't work 
(even I call snap connect - Qt5KeyChain returns an internal error). The migration of HOME data from the deb package to the snap package must be resolved, TQSL is not resolved, I have not tried Hamlib via Serial Port. and much more. 

[QLog_snapv2.zip](https://github.com/user-attachments/files/16015191/QLog_snapv2.zip)

AsciiWolf, 2024-08-02:
Both TQSL and Hamlib serial should be easy to solve. However, the Qt5KeyChain error will probably require more work since it seems to work fine for other Snaps.

AsciiWolf, 2024-09-08:
@foldynl Another possible alternative (at least for now) could be including the deb packages for Ubuntu users directly on the [releases page](https://github.com/foldynl/QLog/releases) just like rpm ones are. Installing packages from PPAs is sadly very (beginner) user unfriendly nowadays.

foldynl, 2024-09-08:
DEB package are prepared for ARM and AMD64 platforms, for all LTS a current version of Ubuntu. It could be 8 packages. I think it will be more complicated than adding the PPA according to the instructions in the README. 

BTW: I was also thinking about APPIMAGE. But you know, there are many ideas, but little time.

AsciiWolf, 2024-09-08:
I think only including x86_64 packages for LTS and previous LTS Ubuntu should be sufficient. Just like it is with the Fedora rpm packages.

AppImage would be great, however it would probably have the same problems the Snap version had. Also, AppImages are [not working properly](https://www.omgubuntu.co.uk/2023/04/appimages-libfuse2-ubuntu-23-04) on Ubuntu 24.04.

AsciiWolf, 2024-11-29:
Were you able to make any progress with the Snap packaging? I am planning to take another look at it in upcoming weeks.

foldynl, 2024-11-29:
no progress since last time. To be honest, it's the lowest priority.

AsciiWolf, 2024-11-29:
Well, PPAs are not a preferred thing in Ubuntu anymore and Snap packaging would probably mean lots of new users. But I get your point. This has not been a big priority for me either. I will try to take a look at it one more time when I actually have more time. :-)

leirbag28, 2025-01-22:
> I think only including x86_64 packages for LTS and previous LTS Ubuntu should be sufficient. Just like it is with the Fedora rpm packages.
>
> AppImage would be great, however it would probably have [...]

I second Appimage. I does not matter that Ubuntu 24.04 has problems with Appimage..that is DEFINITELY a Canonical Problem they need to fix ASAP and I suspect it was on Purpose (by by accident of course) to push SNAP.

Appimages solves sooo many problems especially dependencies and Python and QT issues. I can use Apps that require Newer Python that I do not have nor want installed (because it will break my Older Python apps)

foldynl, 2026-09-19:
I decided not to go the Snap route, but to try the AppImage approach, which in my opinion is more universal.

closed 2026-09-19 by foldynl