foldynl/QLog #398: not direct an issue but a wish and a little bit more

3JWdotcom, 2024-07-03:
Hello,
since yesterday LotW is back online! So i downloaded 0.37 and startet LOTW Importing QSL, so far so good well done. But when i want to upload in the same way tqsl braked up the connection when i had to point out my homeQTH. 
Q: where i had to put the 2 files from TQSL (.tq6&.p12 ....folder in QLOG under Linux?) Hope that works than properly!

Wish: you have got support for FLDigi, super nice...my wish is it would be nice to support FLRIG in the same way you worked with Omnirig
And it will be fine to backup the log automatically to a destination of my choice (ext. NAS/HD/USB) every month/week/day/2h.

mny tks and 73s 
DK3JW Dirk

foldynl, 2024-07-03:
Hello, Unfortunately, you didn't mention whether you are using Flatpak or a package under Linux. However, it doesn't really matter. QLog calls TQSL application, so it is not necessary to have TQSL correctly configured. If you have correctly configured TQSL then just upload your data, and if QLog reports that the appropriate number of QSOs has been uploaded, everything should be fine.

>>>Wish: you have got support for FLDigi, super nice...my wish is it would be nice to support FLRIG in the same way you worked with Omnirig

FLRig can be controlled via Hamlib Interface. Do you have any issue with Hamlib and FLRig ?

>>> And it will be fine to backup the log automatically to a destination of my choice (ext. NAS/HD/USB) every month/week/day/2h.

this is not a task of the QLog application. use CRON or other task "schedulers" for this.

3JWdotcom, 2024-07-03:
hi and thanks for anwering so quickly, I am using Flatpak but unfortunatly i am  a Linux-newbie a typically Windows-dummy from the other world, sorry.
so i try it out later today with TQSL but it is correct working under Gridtracker and Log4OM V2 under Win10 so i thought i had to copy the Certificate File tq6 & p12 to a special directory!

Because i use FLRIG between WSJTx and my Transceiver i haven´t operate with hamlib as Interface seemed to that i am new on this theme as well.

Also i don´t know anything about CRON but thanks for the info, i´ll have a lot to read & learn about linux and QLOG

thnk you 73s

closed 2024-07-04 by 3JWdotcom