foldynl/QLog #402: WSJT-X integration - mode setting

gtorino, 2024-07-06:
During a FT8 QSO and also on registering it, QLog doesn't take Mode information ("FT8") from WSJT-X, but holds that one taken from the rig ("SSB/USB"). I have to manually edit later the registered QSO replacing "SSB" with "FT8". "Accept UDP requests" on WSJT-X is checked. QLog 0.37.0

gtorino, 2024-07-06:
The only way to make QLog recording the correct information is activating "Enable logged contact ADIF broadcast" on WSJT-X:
[image]

So I have:
[image]

But the mode displayed remains uncorrect:
[image]

foldynl, 2024-07-06:
unfortunately I tested now I can't confirm this. The Mode is taken from the WSTJX Log Message and is independent of what the QLog shows. It means that if a request for inserting a QSO comes from WSJTX, the correct mode/submode is saved. The fact that ADIF is on or off does not determine this either. It is necessary not to save QSOs from digital mods using Save QSO in QLog, but using messages from external messages. it is done automatically in the case of WSJTX after every successfully completed QSO.

gtorino, 2024-07-06:
Thanks a lot Ladislav for the very clear explanation. That's also my experience in effect.

closed 2024-07-06 by gtorino