foldynl/QLog #403: QLog not registering QTH field in QSO list

gtorino, 2024-07-06:
QLog apparently is not registering the QTH information:

[image]

On Logbook:

[image]

foldynl, 2024-07-06:
Unfortunately, this is the desired functionality. When you make FT8 QSO with only a 4-digi square, then you can't be sure of anything from the callbook. Only name is parameter that can be used. Therefore, QLog uses only Callbook's Operator name for QSOs that are received from external sources (WSJTX, FLDigi etc).

gtorino, 2024-07-06:
Ok, thank you Ladislav. Sorry to bother, but I still don't get it. Even for FT8, if the qrz.com service is set, it's very likely to have the QTH field filled. Isn't it possible in this case to use the field, if not empty?  Thanks a lot!

foldynl, 2024-07-07:
I'll try to say it differently. QRZ returns all parameters for the given Callsign, including QTH and locator. WSJTX sends Log QSO information. QLog must combine this information and produce a QSO entry in the log. However, it is not possible to combine information that cannot be "obtained" (or validated) directly from a QSO. And that's the case with WSJTX. QSOs made with the help of WSJTX (general digi operations from WSJTX) provide only a limited set of information. Only the Grid is practically usable, which is, however, only 4 characters long.

And now comes the question: QLog has information from QRZ.com and a 4-character grid. How can you be sure that the QTH listed in QRZ.com is the current Callsign's QTH? You simply cannot. Keep in mind that the 4-character grid is a square 110x142 km. The only thing QLog can use from QRZ.com is the name and information that can be derived from the Callsign (DXCC, ITUz, etc.). Unfortunately, this is a disadvantage of WSJTX QSOs.

gtorino, 2024-07-07:
Good point Ladislav, thank you.

closed 2024-07-07 by gtorino

foldynl, 2024-07-23:
Opened poll about this feature #420