foldynl/QLog #404: Cluster spotting not working

muihlinn, 2024-07-06:
Each time I try to spot last QSO it generates something like 

`DX 0 <Callsign> `

which is rejected from the cluster as `frequency 0 not in band`

It happens since I upgraded to 0.37 a few days ago. Built from src in endeavourOS from Arch AUR, as the previous version, which was built the same way and worked just fine.

foldynl, 2024-07-06:
I'm afraid you'll have to provide more information about what QSO you're logging, maybe a screenshot would be good. Was the QSO logged via QLog or is it FT8 traffic?

foldynl, 2024-07-06:
OK, it seems it was an FT8 QSO, wasn't it?

muihlinn, 2024-07-06:
yes, the last QSO made. It's logged just fine (apart the issues commented in #405).

commit 076bf339 (2024-07-10): Fixed #404 - DXC: Fixed zero frequency for last QSO in case of FT8 QSOs
closed 2024-07-10 by foldynl